opentrack/opentrack #117: Not compatible with 64bit games?

Sk1dmarks, 2015-01-07:
I hoped this software would work with the 64 bit version of rFactor 2, but it looks like it does not. Will it be compatible in the future?

sthalik, 2015-01-08:
There's no freetrack version for 64-bit executables. We could compile it for 64-bit if only some apps used it.

closed 2015-03-13 by sthalik

lcambia, 2015-10-09:
Assetto Corsa 1.3 64 bit example.. opentrack not work

sthalik, 2015-10-09:
It should work. Use Process Explorer to see if npclient64.dll is used by the game process.

lcambia, 2015-10-09:
Ti ringrazio, ma non saprei come verificarlo. come posso forzare l'uso di npclient64.dll ?

sthalik, 2015-10-10:
@lcambia you can't. The game process is expected t load it basing on values from registry.

lcambia, 2015-10-10:
We could compile it for 64-bit

sthalik, 2015-10-10:
We don't need to. The game only loads npclient64 and nothing else.

lcambia, 2015-10-10:
ok thanks...I will ask about the AC forum to add to the game a DLL

sthalik, 2015-10-10:
They don't have to. If they support freetrack it's already taken care of. What do they support exactly?