opentrack/opentrack #118: Center/Toggle/Zero keys

Len62, 2015-01-11:
First, I love opentrack and use it exclusively. Is there a way that the "key" options can be set to include the alphanumeric keys as it was in 2.2?

sthalik, 2015-02-01:
The key list is too long that way. Is this really an issue?

Len62, 2015-02-02:
Not really an issue, just was used to the older version. Thank you for your response.

DerangedPineapple, 2015-06-10:
Would it be possible to open this issue (so theres no need to create a new one)? 

It would be beneficial to add a feature where user could designate what ever key he/she wants for Center/Toggle/Zero functions. 

Case example
When in Arma 3 many times when using optics or when zooming in while flying helicopter it would be nice to actually bind a same key to center/toggle that you use for optics in arma 3 so your view especially when zoomed in wouldn't sway all over the place so easily.

Work around I currently use is to use autohotkey and rebind ie. F11 to actually send Return key press but thats kinda stupid in my opinion when it could be integrated into Opentrack in the first place. It would work much more reliable too in that way.

sthalik, 2015-06-10:
@DerangedPineapple it's not a perfect solution as it can get out of sync. It can switch to fast profile when you're starting optics, and smooth profile when exiting optics mode. For instance try pressing escape and pressing the key then.

As for this request and #162 give some time for new Accela binary build gets published. I don't get jumps in optics mode anymore with pure "rotation sensitivity" filtering, without even smoothing or deadzone. I have a solid feeling the flter can be improved enough not to cause problems anymore.

DerangedPineapple, 2015-06-10:
Thansk for reopening this man! Love your work.

I don't understand about switching between fast and smooth profile nor pressing escape etc. Is this somehow connected to binding whatever key/button to center/toggle/zero features?

I merely wanted to support the idea what OP started which was that it would be nice feature to bind whatever key/mouse button to Centre/Toggle/Delay functionalities. That way I could bind whatever button I need for toggle and/or center functionalities so every time when I use optics mode in arma 3 (or zoom in while flying) my tracker would either center headtracking or even toggle it.

You have all the time in the world, no hurry from my side. Opentrack works like a charm allready, these are just nice to have features to make things a bit more flexible and accessible.

sthalik, 2015-06-10:
Actually binding any button makes sense. Software usually allows for it with no button list like ours.

It's a bit of work because of how poorly the keyboard shortcut code is structured. I'd like to get back to it sometime though.

DerangedPineapple, 2015-06-10:
Glad to hear that it would be possible to implement it at some point.

phizz166, 2015-06-13:
+1 to this. Obviously sim-type games/programs are the most likely to use nearly every key on your keyboard...between ArmA and DCS I use pretty much all of the keys that are made available in OpenTrack, even with modifiers. Being able to set this to any key would be best.

Incidentally, attaching a joystick button to center the view would be even better.

sthalik, 2015-07-19:
I agree. We'll also do joystick buttons.

Len62, 2015-07-19:
Perfect! Thank you, will be looking forward to this!

sthalik, 2015-07-25:
Works on all platforms except Windows joystick buttons.

sthalik, 2015-08-06:
How many people want Linux joystick buttons?

sthalik, 2015-11-11:
I don't like the way it's implemented but it has to wait. Hell froze over. Build - https://db.tt/vtxgIy9Z

sthalik, 2015-11-11:
@Len62 @DerangedPineapple please give it a try since I only did light testing. Thanks.

Len62, 2015-11-11:
Gave it a try, 1st, when you select bind, and press a button on the joystick you have to hold the button down until the window auto closes, shouldn't it just register it on click/release? (Like the keyboard inputs) 2nd, when you hit ok to close the option window the joystick assignments do not hold, they all change to button "15". 3rd, unlike in the joystick tracker, were it identifies the 3 controls I have, joystick, throttle, and rudder pedals, there is no option in the shortcut bindings to select the device you want to use.

sthalik, 2015-11-11:
It detects a stick only after a 2-second delay since startup due to a bug.

Is it that button 15 is always pressed on your device, like a mode switch or something?

You SHOULD be able to select the device, merely the tooltip won't show that information.

Len62, 2015-11-11:
Button 15 is just a toggle on the throttle controller and only active when pressed. I noticed that none of the buttons on the throttle are recognized if I try and set a short-cut. Buttons on the joystick are sometimes recognized and revert to 15 after clicking OK.

Len62, 2015-11-11:
If it helps any my rudder pedals are ID'd as joystick 0, joystick itself is joystick 1, and throttle is joystick 2. These are the controller id's assigned by windows. They do show up correctly identified in the joystick tracker though.

sthalik, 2015-11-11:
Recheck the current one - https://db.tt/oLrcqfz7

I'll hook you up with a debug build if it's still porked.

Len62, 2015-11-11:
Okay! I was able to correctly assign shortcuts on both the throttle and the joystick, and the settings are being retained. Center, zero, and toggle all working. I did notice that the zero and toggle only work while the button is being held down, not sure it that is by design or not. Using keyboard shortcuts if like having off/on switch.

sthalik, 2015-11-11:
It's meant to be the same toggle, bug fixed - https://db.tt/FDlMX9CD

Does it work for all of your USB devices?

Len62, 2015-11-11:
One last thing. Could the ctrl, alt, shft modifiers be set to work in conjunction with the joystick inputs? Thank you for all of your work on getting this implemented! I always recommend opentrack to anyone looking into head tracking options! Just saw the new build, will test it.

Len62, 2015-11-11:
Tested the new build. Input from both the throttle and joystick are working perfectly! Only other USB device I have are the rudder pedals.

sthalik, 2015-11-11:
Keyboard shortcuts should work now - https://db.tt/W27DR8k1

sthalik, 2015-11-11:
To be clear - we don't use joystick indices in Windows to distinguish them. We use unique joystick identifiers. It should even work when switching USB ports, at least in theory.

sthalik, 2015-11-11:
@Len62 can we get this checked and next rc out of the door today? Thanks!

Len62, 2015-11-11:
Tested the new build and everything is working great! Using the joy identifiers is the way to go otherwise windows can muck-up usb assignments.

sthalik, 2015-11-11:
I agree. They just won't show in the UI since these identifiers are pretty long. I don't want to resize the shortcut dialog for for their sake.

Len62, 2015-11-11:
It's fine as it is. And you are right, some of those identifiers are way too long. Thanks again!

sthalik, 2015-11-16:
We're good, and will reopen if more bugs.

closed 2015-11-16 by sthalik

Boberro, 2016-03-27:
Hi, I'm using windows version of 2.3-rc21p11 and it's impossible to bind mouse buttons for hotkeys (nothing happens when I click). Joystick seems to be not working too, but I have Steam Controller so it's strange by itself.

Len62, 2016-03-27:
I don't think the hotkeys can be bound to the mouse. Joystick bindings work fine. Are you running opentrack w/admin privileges? What joystick are you using? I'm not familiar with the Steam Controller, maybe someone else will comment.

Boberro, 2016-03-28:
Lack of possibility to bind mouse buttons for hotkeys is a shame, as today standard mouse has 5 buttons (thumb "back" and "forward" seem to be standard now (and I definitely use them for tracking toggle and center with FacetrackNoIR)

sthalik, 2016-03-28:
Mouse bindings are a separate method of input that would have to be implemented. Code is welcome.