opentrack/opentrack #120: Mouse emulation

jorgen41, 2015-01-14:
When mouse emulation is selected the mouse just centers and freezes. cant move pionter either in windows or ingame.

Tracking works in Opentrack..

Using:
Running Opentrack with Admin
windows 8 64bit
Opentrack 2.3-rc8
freepie IMU in android 4.4.2

mm0zct, 2015-01-14:
This is related to another recent bug/feature change request I'm investigating when I get a minute.

Basically OpenTrack uses absolute mouse emulation - every tracker update it puts the mouse where it thinks it should be, and moves this based upon the sensor input (be it face track, oculus rift, point track etc.). This has the side effect of over-riding all other mouse input. I'm hoping to switch the mouse to a relative positioning based system, which is what most people expect it to do (acting like those gyro/accelerometer based air-mice you can buy).

jorgen41, 2015-01-14:
Exacly what i wnted :-) Changing the behavior would open up Opentrack to a pletora of games. I have tryd with like wifi mouse on my phone and the likes and it works kind of well.

matskatsaba, 2015-02-02:
further problem with mouse emulation
Windows7 64bit
Opentrack2.3-rc9
Tracker: Oculus Rift - HMD
Protocol: Mouse emulation

In mapping window /Options: disable all,except Yaw=Yaw; Pitch=Pitch[Invert]
In Mouse emulation options: Map mouse X to Yaw, Map mouse x to Pitch

When I hit start, it works properly, mouse moves properly to the point I look (more or less).
When I started a game (gta3 san andreas), the mouse pointer was working properly in the menu, but in-game where it`s hooked to camera control, so the camera spins around unless the oculus is at center.
Might this be game-sided?

sthalik, 2015-02-04:
It's known, will be fixed before release.

sthalik, 2015-03-13:
Duplicate of #113

closed 2015-03-13 by sthalik