opentrack/opentrack #121: Add a Start/Stop key binding for completely resetting the tracker

hduregger, 2015-01-16:
I would like to propose the addition of a key binding to the _Start/Stop_ functionality of the tracker to allow completely reset tracking from inside a game. I can alt-tab out of the game, _Stop_ then _Start_ and alt-tab back into the game. But this is of course no good solution when flying a helicopter.

A few reasons

* In only a few seconds of virtually looking back over the shoulder in the DCS flight simulation and looking forward again, the tracking area becomes misaligned. I can _center_, but the tracking area sometimes drifts a bit too on my face. Which becomes worse over time (few seconds of action).
* Sometimes it even loses my face and starts to track the white (but structured) wall behind me :D
* When I scratch my face, the tracking area often becomes misaligned

This is all a bit weird though, because I can move my hand over my face and it is pretty robust at tracking and reaquiring the correct tracking area. Still the problems occur in action.

If I remember correctly _Toggling_ in facetracknoir (and trackir?) does reset (_Start/Stop_) the tracker completely. Not just stopping the data stream. Hence, you could reset tracking if something became misaligned/misidentified.

Using 
* Windows binary release 2.3 rc8
* ht -- face tracker
* Accela filter
* freetrack 2.0 Enhanced
* PS3 Eye camera
* starting out with error ~3, keypoints ~100 (if that is of any help)

Apart from these issues this is a much better experience than the last facetracknoir I tried. Thanks and congratulations to this great program!

Best regards,

Helmut

KyokushinPL, 2015-01-18:
You have Keys tab and there you can add button for stop/start tracking, center headtracking and reset.

hduregger, 2015-01-18:
As far as I can see

* Center: zeroes the game data once to a new reference value and keeps updating the game data
* Toggle: toggles between 
  * freeze the current game data
  * update the game data
* Zero: toggles between
  * freeze zero as the current game data
  * update the game data

The current three options don't reset the tracker, they only influence the game data.

As far as I remember in facetracknoir the option for Toggle would also reset the tracker completely, e.g. the face tracker would reacquire your face. Or maybe I don't remember it correctly but you could fix tracking issues with it (e.g. it started to track something different from your face). This only happens by pressing Stop then Start here.

Hence, I would like to proposes either changing the Toggle behaviour here or introducing a single new toggling key binding for the current Start/Stop buttons (which already has toggling functionality).

The reason is that you can obviously use the key bindings from inside a game, but you have to alt-tab in and out of it to get access to the Start/Stop buttons.

hduregger, 2015-01-21:
In facetracknoir 1.7 the shortcut key option for "Start/stop" has the additional options to "Zero" or "Stay" the view when off and to "Stop tracking" or "Keep tracking" when off.

[image]

Sidenote:

Looks like my judgement was too fast. I tried facetracknoir 1.7 again, and I seem to get a better results there. Additional to the problem I stated originally, it seems as if re-centering is easier with facetracknoir. But maybe it is all just the settings. Don't know.

I was initially very confused about what the purpose of the two projects are. I thought opentracker is a continuation of facetracknoir somehow or improvement. Now I think its reason is the multi-platform support. This is great. I am a Linux fan myself, but as the circumstances are, I have to use Windows for certain applications.

The issue still stands, I am sure it would be helpful to add the shortcut functionality. Maybe design it more after facetracknoir, although I admit their GUI can be confusing.

sthalik, 2015-02-01:
I'd like to avoid doing that for a malfunctioning tracker. Ideally it's the tracker that needs fixing and not workarounds like this.

I've pushed a change to the face tracker, so that extreme poses that are misdetected don't lead to tracking breaking down, along with better mesh position.

But the truth is face tracking is faulty and relying on it for combat flying is just pain.

sthalik, 2015-02-17:
The HT face tracker was updated in time for rc9. It should reset on failure faster now.

closed 2015-03-13 by sthalik