opentrack/opentrack #122: Gain/exposure setting in PointTracker tracker

prisonier, 2015-01-21:
In FreeTrack 2.2 (the soft, not the tracker) we was able to set the exposure/gain directly from the GUI. Now i have to do it in the driver software of my PS3 eye. The driver i use is code laboratories.

cheers !

forteh, 2015-01-22:
That shouldn't really be a problem though, I set my cl test to minimum gain, maximum exposure and opentrack works perfectly with the 3point ir clip.  I've never altered the camera settings, the above gives great point extraction and with a piece of floppy disk film over the lens as a visible light filter I can use it in any light conditions I want :)

prisonier, 2015-01-22:
Hello !

I don't have a floppy disk filter, i'm using a film. 

" I set my cl test to minimum gain, maximum exposure" 
so you had to change it too :)

Now i've understood that i can change those settings in the CL driver but it took me more than 1 hour (yes i'm very dumb sometimes). I'm sure that centralizing all settings in 1 GUI is a good practice and will make new comers life easier.

forteh, 2015-01-22:
I think opentrack works just fine with it set to the default auto settings anyway, I just like to fiddle and mess with stuff; hence the home built head tracking, the wall mounted overclocked pc, gpu cooler hacked from a previous generation card and usb converted tm rcs rudder pedals :D

I suppose if the camera was going to be used for multiple functions then it might be worth having application specific settings (much like a game controller) but for dedicated hardware such as headtracking (it should be dedicated imho) it's another layer of complexity in the software that could potentially cause issues :)

prisonier, 2015-01-24:
The camera is dedicated for OpenTrack and it did not work with default settings.

I don't realize how much it is complicated, i juste suggest it because it was in FreeTrack settings. :)

Anyway, everything is working nice today

closed 2015-03-13 by sthalik