opentrack/opentrack #125: DualShock support

brndd, 2015-01-27:
The PlayStation 3 and 4 controllers have gyros and accelerometers on them. At least with the PS4 controller, these even seem to work on PC.

Is there a way to make these work in Opentrack? The third-party driver (DS4Windows or something) I'm using with the DualShock4 can access the gyro/accelerometer data and even map it to various buttons, but there are not enough axis available to map all of the six axis so currently I couldn't get it to work.

I've seen a video of headtracking with a smartphone, so I was hoping the same could be done by strapping my PS4 controller to my head.

KyokushinPL, 2015-01-27:
Good idea, but i am curious about weight of gamepad. Some guy on Elite Dangerous forum tried to use ps3 gamepad for tracking and he said it is too heavy to be long term useful.

sthalik, 2015-02-01:
Ideally "DS4Windows" would fuse the sensors to produce rotation data.

Including a device driver in opentrack is wrong from software engineering perspective. It should be its own project.

closed 2015-03-13 by sthalik