opentrack/opentrack #126: Wrong calculation of 3 Point Cap

clawhammer64, 2015-01-28:
Every time i download new version and check how my 3 LED Cap model works.
But always i see same issue. The red line on the picture - thats the stuck for Opentrack, after which begins wrong calculation.

If the original Freetrack software continue YAW without any problems, the opentrack instead make some ROLL or etc. after crossing red line.

And the second question - why from version to version  i see different output results if the PointTracker Plugin remains the same?
[image]

alowva, 2015-01-28:
duplicate issue, posted here a few weeks ago: https://github.com/opentrack/opentrack/issues/112

(not sure if theres a way I can report this as duplicate, so sorry if i have posted redundantly)

clawhammer64, 2015-01-28:
Thanks! I'm just remind about it cause new versions have not any changes.

sthalik, 2015-02-01:
Results are different since code now takes camera field of view into account. Due to this translating doesn't affect rotation.

sthalik, 2015-02-01:
Closing, notify if dynamic pose resolution doesn't work.

closed 2015-02-01 by sthalik

clawhammer64, 2015-02-04:
It doesnt work, already tested on RC9. Results are same.

alowva, 2015-02-04:
works fine for me after the update

clawhammer64, 2015-02-04:
Write me model dimensions please, i'll recheck.