opentrack/opentrack #127: OpenTrack affecting keybindings after client & game both running in admin mode

thegreyfawkes, 2015-01-28:
Hello!

I eventually was able to set up tracking to be (mostly) where I wanted it to be. Once starting Elite Dangerous, though, some custom keybinds to my X52 flight stick are no longer being registered. (i.e. lower hat on flight stick set to up/down/left/right thrusters). The strange thing is...I am able to enter controls in ED and bind the keys again - so the game is recognizing that it's being pressed. Back in the game however, no thrusters are being used when I'm moving the hat on the flight stick around.

I have verified that I am running OpenTrack in administrator mode, as well as the Elite Dangerous client itself (not the shortcut). Most of my google searches for this issue keep coming back to "common issues" thread about the keybindings and admin mode.

Anything that can be provided would be greatly appreciated. I am eager to enjoy the full experience of using OpenTrack as my cheap-but-effective head tracker solution!

KyokushinPL, 2015-01-28:
Strange. I have x52 pro and it works without any problems.

thegreyfawkes, 2015-01-28:
That's exactly what I've been doing - no profiles, just binding the keys directly into the game. My setup is running from a recently reformatted SDD, so I have had zero issues with drivers this time around. What really gets me is that the keys are bindable in settings > controls...just actually USING them while playing ED with OpenTrack suddenly makes it not work.

I'm going to try switching to various USB ports for the X52 when I get home today and see if that changes anything, but I remain doubtful.

Would you be open to sharing your OpenTrack settings so that I can compare them to mine?

sthalik, 2015-02-01:
For clarity, you don't need to run the game in admin mode. For keyboard shortcuts in opentrack, it's necessary to run in admin mode when the game does.

opentrack doesn't touch joystick input, unless you're using a joystick tracking source. Even then, this is a "background" mapping of the joystick, allowing other apps to use it too.

A far shot but you might wanna see if you're using "vjoy" or similar software and uninstall it.

closed 2015-03-13 by sthalik