opentrack/opentrack #128: Center issue in opentrack-2.3 release candidate 9

Stefan-deHoDev, 2015-02-01:
The following change in release candidate 9 interferes with apps/devices that have inbuild center features:
- centering will perform implicitly on first pose received

The problem with this new feature is, the app/device has no way of knowing what opentrack perceives as the center.

My proposed solution:
In Mapping have a checkbox to enable/disable this feature

sthalik, 2015-02-04:
Recentering works as usual, there's no loss of functionality. Also, the first pose received is useless without any center applied, as translation values are high.

Stefan-deHoDev, 2015-02-04:
No, you didn't understand the problem. You are right, recentering in opentrack works as usual, but I'm not taking about that!
My own app "Headtrackr" has it's own recentering feature. It asumes that the programm receiving the data uses yaw = 0, roll = 0, pitch = 0 as the center. Now in my app you can, for example, tap the screen and your current looking vector is mapped as 0.

Now with opentrack 2.3 release candidate 9 the first position received gets set as center. So it's not yaw = 0, roll = 0, pitch = 0 anymore.
It's impossible for my app to know what opentrack has as set as center, making the recenter feature in MY app useless.

I'm not sure what you mean with your second sentence. Everything before rc9 works great, because it uses 0, 0, 0 as it's default center position. 
With rc9 the center position the app uses can not be predicted.

FlyingCircus-, 2015-02-06:
I'm not sure I understand the issue. I believe the output values of OpenTrack are unlikely (0, 0, 0) with or without the automatic centering on the first pose (unless you're on that pose by a very fortunate chance). However, automatic centering on the first pose isn't working right according to my experience with RC9 (see #106), so whatever is wrong with the first centering could also be causing your trouble.

closed 2015-02-17 by sthalik