opentrack/opentrack #129: Smartphone head tracking Suddenly stopped

Ronan6793, 2015-02-02:
So I have been using this with my Samsung galaxy S5 for head tracking in Arma 2 and War thunder for some time now and all was well until today when I loaded it up and pressed start it doesn't work and I have tried everything from re install to trying my forwarding still nothing. All of my setting are the same as they were yesterday but today it just doesn't work, this is the screen I'm getting now http://gyazo.com/879bc6245f4f29e1dd2af58cec558593

Stefan-deHoDev, 2015-02-03:
Show us a screenshot of your mapping settings (options).
I had a similar problem yesterday, where in "output remap" yaw, pitch, roll, x, y, z were all mapped to x.

Ronan6793, 2015-02-03:
Here http://gyazo.com/63fbf2c60958224aa9e2ffe70869ab32

sthalik, 2015-02-04:
On the screenshot the start button isn't pressed. Could it be the cause?

Ronan6793, 2015-02-04:
"until today when I loaded it up and pressed start it doesn't work" I know it isn't pressed in the screenshot because I had already tryed

Ronan6793, 2015-02-14:
Case closed dont know how it fixed but Its working now

closed 2015-02-14 by Ronan6793