opentrack/opentrack #130: Issue with registry entry conflicting between pre-releases

Stefan-deHoDev, 2015-02-03:
I'm testing multiple opentrack 2.3 release candidates. I'm facing the problem that the path for the standard setting file is saved in registry. This means that when configuring opentrack 2.3 rc5 and saving it and then starting opentrack 2.3 rc9, rc9 loads the settings from the path that was saved in rc5.
This gets even worse when rc5 was deleted, because then the path does not exist anymore and weird standard settings are selected (all inputs mapped to translate x output...)

Following proposal:
Don't use the registry, save a "defaultconfig.ini" or "default.config" in the same directory as opentrack.exe, that points to the settings ini file saved in the settings subfolder.

This would allow opentrack to be fully portable!

sthalik, 2015-02-04:
Issue with storing settings in program's directory is when deleting the whole directory.

When you run opentrack the first time, the path stored will be the pathname to where opentrack was run from. Though that depends on drive letter and exact path.

I'll consider using software's current directory as long as it's the one used, so let's leave this ticket open.

FlyingCircus-, 2015-02-06:
I must say I agree with this suggestion. I experienced the same problems when switching between versions,

closed 2015-02-17 by sthalik