opentrack/opentrack #136: Opentrack freeze upon game launch

goshops, 2015-02-16:
For a while I've been using Opentrack in F1 2014 in the Oculus Rift, however yesterday it suddenly stopped working. Previously I would launch Opentrack, then Vorpx, then F1 and would have no issues. However now when F1 launches it causes Opentrack to freeze which disables headtracking. Running the game in windowed mode however allows some head tracking functionality but it is so laggy that it is unplayable. Any ideas or help? There have been no changes to my system, no updates, no new installs, no new downloads, no crash report, etc.

System info:
Opentrack 2.3-rc9
vorpx 0.7.5
Win 7 64
i7 4790k
GTX 780ti

sthalik, 2015-02-17:
Not without debug info which is probably overkill to supply.

Have you tried rebooting though?

goshops, 2015-02-17:
I figured it out, reinstalling the Oculus runtime resolved my issue.

closed 2015-02-17 by sthalik