opentrack/opentrack #137: Newbie How-To?

OlHall, 2015-02-26:
Is there any sort of document that describes the process of setting up/tuning head tracking using OpenTrack from a newbie's point of view?

Context: I'm using point tracker, a homemade 3 point clip and a doctored PS3 Eye Camera with the CL Eye Driver (and playing DCS).

I "get" how it's all supposed to work, but there seem to be some anomalies in OpenTrack that make me question what's going on.

For example: If my mapping is set to "1:1" and I'm not using any filter algorithm, why isn't the raw tracker data identical to the game data?  I have centre pose all set to 0, and disabled all compensation... so this seems wrong.

Secondly, I have a peculiar setup because my camera is not in the same plane as my primary display (my simpit means the camera is mounted 30deg below the centre line of the screen).  I assumed that the camera offset for pitch would enable me to compensate for this, but it certainly doesn't work.  If I move my head forward without moving up or down, the TY value increase - as you would expect if there was no compensation for the tracker's projection (since my IR dots are moving UP the screen as I move forward).  Since I would have thought this compensation would *have* to take place in the tracker model plugin, I'm assuming that I can't accomplish what I'm trying to, in which case - what does the camera offset actually do?

Finally (for now!) I think I understand the point of the centre pose adjustment (set statically, nothing dynamic) - correctly me if I'm wrong - this is to allow you to make allowances for the orientation/translated position of the tracker plugin's output vs your physical environment, such as the fact that my headclip isn't perfectly vertical when my head is?  However, the mapping graphs display RAW value, rather than the RAW value *after* the centre pose adjustment.  I may well have the wrong end of the stick here, but that's why I think a document outlining the "processing pipeline", and a best practice of how to set things up would be really helpful.

I hope that my camera offset thing isn't going to mean that my set-up will never work, but thanks for a cool bit of software.

sthalik, 2015-03-13:
For the first question, in "Keys" menu disable "center in startup" to hopefully get same raw/game data again.

Off-center camera fixed with commit. Will be in next RC.

Documenting the pipeline is pointless since it's subject to change. Right now it's done before filtering, to give the filter information on how sensitive user's axes are.

Adding that detail to mapping screen is only clutter so it won't be implemented.

closed 2015-03-13 by sthalik

OlHall, 2015-03-14:
Wow...  You probably don't realise it, but you've just helped make my decision to go and spend money on a TrackIR.