opentrack/opentrack #139: Hatire Arduino plugin

huliqan, 2015-03-05:
Please help to add Hatire Arduino plugin of FaceTrackNoIR in Opentrack
http://sourceforge.net/projects/hatire/

sthalik, 2015-03-13:
If someone submits working and tested code, then sure. I have no device, so no means of getting it done by myself.

huliqan, 2015-03-14:
I use this plugin with Arduino nano 3 + GY-85 ,and This kit works on a lot of smoother and more accurate than diode frame and Clip! But I have to translate the data out from FaceTrackNoIR to OpenTrack through Vjoy protocol because only it gives me a full review of the overlook of the pitch in BoS (IL-2 Sturmovik Battle of Stalingrad)

zakamik, 2015-03-16:
It would be great If anyone could suggest any samples or stubs for new plugin.

sthalik, 2015-03-31:
Merged in a449babd2d3bd46c0aafe0ed0109a379d65c94f3.

Please check if https://www.dropbox.com/s/ok73ddag92ciym0/opentrack-2.3-PLUS-hatire.7z works. I won't maintain the code long without confirming it works.

zakamik, 2015-03-31:
Doesn't work at all, sorry. Tried different ways, without luck. No digits seem to come in.

sthalik, 2015-03-31:
There's a status field in tracker settings that shows debug information. It could know what the problem is.

zakamik, 2015-03-31:
Status field shows:
tps: 0
Info: HAT START

Will try another sensors later.

FlyingCircus-, 2015-03-31:
@huliqan I'm curious about your statement about the inertial tracker. Last time I checked I got the impression they were less accurate than the LED clip tracking. Are you getting such good results? Do you emulate not only rotations, but also translations? Does it get an increasing offset if it's not centered from time to time? Not sure this is the place to ask these questions...

sthalik, 2015-03-31:
Inertial trackers with no other support don't do translation, no.

Close proximity to magnets ruins the device's operation. How much distance from headphones is necessary, can someone share? Is it bearable to use with headphones?

Drift for a particular axis can be offset by a very mild drift toward center in the tracking software. It's unnoticeable with the hardware's precision.

Level of accuracy of LEDs and inertial trackers is high enough that the difference isn't perceivable. I don't think you can spot 30/60/120 FPS difference with LEDs too.

zakamik, 2015-03-31:
Headphones mounting is possible but depends on the make. Sometimes moving sensor away for 1-2cm removes any influence.

Drift is mostly a product of improper device calibration caused by temperature variation, multiplied by accumulated errors in calculation (floating point). Can be removed or greatly reduced by filtering with side effect of decreased sensitivity.

Difference in accuracy of LEDs and inertials is unnoticeable, however UX with properly tuned inertial tracker is much better than with LEDs (excluding missing side movements).

FlyingCircus-, 2015-03-31:
Thanks for your input guys. The lack of translations is a pitty, I'll stick with my LEDs for the moment.

huliqan, 2015-03-31:
The LED-Clips diode have much worse response in comparison with hatire(hat arduino tracker)!
I completely abandoned the diode clips after tried HATIRE!

huliqan, 2015-03-31:
<<Please check if https://www.dropbox.com/s/ok73ddag92ciym0/opentrack-2.3-PLUS-hatire.7z works. I won't maintain the code long without confirming it works.>>
it work ! but just twitches squid-frame

huliqan, 2015-03-31:
it work strangely,only if only if the redirect yaw pitch roll to shift X Y Z

huliqan, 2015-03-31:
I use
void FTData()
{
  if(Serial.available() > 0)  serialEvent();
//    roll=rollAngle;
//    yaw=panAngle;  
//    pitch=tiltAngle;
  yaw=panAngle;
  pitch=tiltAngle;
  roll=rollAngle;
  hat.gyro[0]=yaw-XZero;
  hat.gyro[1]=pitch-YZero;
  hat.gyro[2]=roll-ZZero;
  hat.acc[0]=hat.gyro[0];
  hat.acc[1]=hat.gyro[1];
  hat.acc[2]=hat.gyro[2];
  
     Serial.write((byte*)&hat,30);
  hat.Cpt++;
  if (hat.Cpt>999) {
    hat.Cpt=0;
  }
  delay(10); 

}

sthalik, 2015-04-01:
Please recheck <<https://www.dropbox.com/s/ok73ddag92ciym0/opentrack-2.3-PLUS-hatire.7z>> it should work for angles now.

huliqan, 2015-04-01:
I will test few later. Thanks!!!

huliqan, 2015-04-01:
No,as was - work only side shift redirect  and doesn't remove curves-point (yelow from screen-shot)
[image]
[image]
[image]

huliqan, 2015-04-01:
my test video
https://youtu.be/dNvYSdWc91A

huliqan, 2015-04-02:
sorry for the "doesn't remove curves-point (yelow from screen-shot)" it work good! I jast forgot stretch curves window:)))

zakamik, 2015-04-02:
Doesn't work with my device at all, unfortunately.

sthalik, 2015-04-02:
it's a bug that mapping window wasn't forced to exact size. Will be fixed for rc10.

@huliqan setting sensitivity to max gives you more filtering, you shouldn't need more than 1/5 on that scale with an inertial device.

sthalik, 2015-04-02:
@zakamik if it works on facetracknoir, it'll work here. You could've misconfigured it.

huliqan, 2015-04-02:
in zakamik does not work, because in firmware not have redirect 
  hat.acc[0]=hat.gyro[0];
  hat.acc[1]=hat.gyro[1];
  hat.acc[2]=hat.gyro[2];
probably
with setting sensitivity in new Accela-filter I've already figured out, thanks

huliqan, 2015-04-02:
night shooting video work hatira in BoS

zakamik, 2015-04-02:
@sthalik I could have, but I haven't. Tried it on two different trackers - GY85 and GY86. Both successfully tested on FTNoIR. First one is silent dead, second randomly flashes with input data, without any practical result. Will try to make a video later.

huliqan, 2015-04-03:
my test videos:
1.opentrack -2.3rc9-30-hatire
2.opentrack -2.3rc9-32-hatire
turns incomplete and don't work key-center
3.opentrack -2.3rc9-32-hatire(redirect yaw>X>yaw & pitch>Y>pitch)
it's works much better:))) but the range of turnings limited program
https://youtu.be/9h18tUJLEUc
https://youtu.be/HwVAKVbg8tU
https://youtu.be/U6GKDpa8zsE

huliqan, 2015-04-03:
And were in ver.2.3 settings deadband (dynamic dead zone
[image]
)???

sthalik, 2015-04-03:
@huliqan deadband has been integrated into sensitivity setting. What's with the turn range though, is this a general issue?

@zakamik Are you able to provide standard error output on the current build? If not, there needs to be a build that opens a console window for that to be easily visible.

zakamik, 2015-04-03:
@sthalik if you could give me a hint how can I collect that stderr output - when I start app in console, it simply detaches.

sthalik, 2015-04-03:
@zakamik it works when launched in Cygwin terminal, or as Visual Studio attached debugger.

huliqan, 2015-04-03:
"What's with the turn range though, is this a general issue?"
I do not know what is the problem, but in FTnoA does not have this issue,and if use FTnoA(hatire)>VJoy>opentrack(joystick)>freetrack :>>does not have this issue too

sthalik, 2015-04-04:
@huliqan I need more info on that. What's the max raw value you can get on yaw? Does it depend on which axis is redirected into yaw?

huliqan, 2015-04-04:
Ok,friend:
if like this
[image]
max raw value on yaw = 64-67 on pitch = 64-67
[image]
if redirect axis to shift like this
[image]
max raw value on yaw = 99 on pitch = 99
[image]
[image]
[image]

huliqan, 2015-04-05:
hatire redirect axis to shift
https://youtu.be/BhJ5k4sPHKU

zakamik, 2015-04-07:
@sthalik I can't install cygwin for some reason, tried in winavr shell, got no output.

doveman, 2015-04-08:
I'm using a MPU-6050 sensor and a Pro Micro, feeding a HC-05 Bluetooth module, all powered by a rechargable USB battery and mounted on the top of my headphones. I haven't noticed any magnetic interference from the headphones.

Whilst it sometimes works in FTNoIR, it has a habit of ceasing to produce any output, even though the input is still showing as receiving data, so I presume that's a bug with the plugin rather than the hardware/firmware. I previously used it with a wired USB cable instead of BT and that had some issues, particularly with the COM port frequently changing and difficulty getting the driver to stay working, so using BT eliminates those problems and makes it a bit easier. This should be the code I'm currently running https://drive.google.com/a/stoptheviolence.co.uk/file/d/0B1fDI89phEESTDAyb3k1X2hhWkE/view

I haven't tried it in OpenTrack yet as I wasn't aware there was a Hatire plugin until today but I'll give it a go when I get a chance. 

I wonder if anyone's tried using the Razer 9-DOF IMU or Sensor Stick? https://github.com/ptrbrtz/razor-9dof-ahrs/wiki/Tutorial Despite it's name, I'd be surprised if it provides any more than the 3-DOF we get with the other sensors we're using.

doveman, 2015-04-08:
Using rc9-32 and the same Accela settings as shown in huliqan's post from 6 days ago, the raw data is alternating between a + or - value and 0 constantly and consequently the game data is all over the place.

I only have Yaw and Pitch mapped to RotX and RotY respectively but still see game data for roll. I can fix this though by disabling Roll under Mapping - Output remap, so it's not a problem. 

Testing by clicking Start without first turning on my device, I'm happy to say that it simply reports that the COM port is unavailable, whereas FTNoIR crashes completely.

On a general point, it would be nice if the Mapping screen would stick to the size it's dragged to, as currently it resets back to the default size when opened and can't be resized on the Options tab it opens to. In fact, it would also be nice if it remembered which tab was open when the windows was closed and opens to that next time. These changes would make it a lot quicker and easier when going back to tweak the mappings.

huliqan, 2015-04-08:
""I haven't noticed any magnetic interference from the headphones.""
magnetic interference may be present only on the sensor with magnetometers (MPU9150, GY-85 ...)

doveman, 2015-04-09:
Doh! To be honest, I'm not even sure whether I'm currently using the MPU-6050 or the GY-85 as I bought both (actually two of the GY-85) and tested with both at various times, so I can't remember which one is currently soldered to my board.

Is there any real advantage to having the magnetometer rather than just the accelerometer and gyro?

huliqan, 2015-04-10:
with  magnetometer no drift to  X-axis and it is not necessary to calibrate!

doveman, 2015-04-10:
I see, thanks.

sthalik, 2015-04-10:
@doveman there was a bug with angle scale depending on whether they were inverted in hatire. It's fixed now, same address for the build as before.

huliqan, 2015-04-12:
in new build hatire stopped working completely ,if yaw-to-yaw and pitch-to-pitch.Work only redirect yaw-to-X and pitch-to-Y

MartinJG, 2015-04-19:
Hi Guys, I'm also trying to use the hatire protocol and have similar issues to huliqan. I also seem to have very jerky gathering of data whereas in facetracknoir it is silky smooth.

Anyway, @sthalik, I am more than willing to work on the hatire plugin to try and fix the issues as I have my own Arduino setup. My issue is building the plugin. I have QT, but I can never seem to get a compatible plugin to build. I don;t want to have to build the whole opentrack project because I don't want to bring in all the other dependencies. 

Can you give me a heads up on how to build just a compatible plugin and I'll get to work. If I genuinely do have to build the whole lot, then please let me know what I need and how you have it setup as I am poking in the dark here ;)

Thanks for your help.

huliqan, 2015-04-19:
and I want to add old filter Accela from 2.2 ver.  to 2.3 ver. opentrack
How it's posible?

sthalik, 2015-04-19:
@huliqan it's back to where it was before in the current build. Max 65 degrees isn't something obvious in code, you could adjust it with mapping.

@MartinJG actually Qt is the only required dependency, others are optional.

doveman, 2015-04-19:
@sthalik Seems my reply disappeared but I tried the new build and now there's no Raw tracker or Game data at all. In the Hatire settings I see tps fluctuating anywhere between 45 and 72 but that's it.

MartinJG, 2015-04-20:
@sthalik thanks for the heads up, good to know. Unfortunately, when I run cmake to try and create a project it gets very upset. It even complains about my QT setup claiming it can't find it. It also complains about the face tracking dependencies wtc etc. Any chance you can give me some pointers to get it to build? I can then try and track down the illusory issues. 

I haven't had a chance to test your latest build yet so I'll give that a shot tonight my time and feed back.

MartinJG, 2015-04-24:
Hi, had a once in a decade storm here in NSW Australia, so been without power for the last 3 days. Back on now so I'll have a good go at compiling the plugin this weekend and see if I can trace any of the issues we are having.

huliqan, 2015-04-24:
Hi,Martin!
good luck in the work on the plugin

MartinJG, 2015-04-25:
I have had many issues so far lol. First I couldn't compile becasue my compiler is a M$ one and they don't support some of the latest c language revisions. Tweaked the code to get round that. Linked it, go tmy .dll and now it won't run. I think it's cause I compiled against Qt 5.4.1 and this project may use an older Qt version. Gonna search for which exact version this uses. If anyone can tell me now that would really save me some time lol!!

MartinJG, 2015-04-25:
Ok, ignore my last pose. Discovered that opentrack isnt really ever going to work with visual studio. Not without too much work. Using MinGW which I guess is what shalik was using. Is that right?

Anyway, after some more blind stabbing in the dark I got it to build. I fixed the first problem I was having with it flicking all the values to 0 every second or so. Is that an issue we all shared (looing at the code we should have done!!). So my values are all smooth now and I for one am much happier.

@sthalik, is there a preferred way of getting my modifications back in your repository?

for those wanting to test to see if I made things better, you can get the binary here:
https://github.com/MartinJG/OpenTrackHatireTest/tree/master/WindowsDll
Let me know how it goes.

doveman, 2015-04-25:
@MartinJG, thanks that's fixed my problem of having no data at all, let alone flickering back to 0. With RotX mapped to Yaw, RotY mapped to Pitch and RotZ mapped to Roll in the Hatire settings (which is how I had it in FTNoIR) and Yaw, Pitch and Roll mapped to Yaw, Pitch and Roll in Output remap, it's working great now.

Also, if I unmap RotZ from Roll, I no longer need to also unmap Roll in Output remap to prevent any roll data being seen.

huliqan, 2015-04-25:
@MartinJG, your plugin is realy working!!!First test is very good!!!
Thanks!!!

huliqan, 2015-04-25:
a little fly in BoS and feelings of working your plugin-perfect!

MartinJG, 2015-04-25:
Glad that it seems to be better. I think all the weird effects had the same reasonably simple cause. I still have a minor issue with getting command timeouts even though they work fine. I think it is caused by the way a qt library function waits. Not an issue really, just irritates me lol!, I was thinking of extending the protocol so you send of gyro, a cel and magnetometer data and then I'd use rtimulib to do sensor fusion in the hat ire plugin. I'd keep the existing way of working too of course. Let me know if you think that's has a use for you?

huliqan, 2015-04-25:
I tested this plugin in BoS and DCS?and I  did not notice timeouts.
What does this mean?I have everything works great!
sensor fusion accelleration data+gyro in the hat ire plugin-the best idea!Very need!
Thank you very much!!!

MartinJG, 2015-04-25:
So my arduino implementation allows me to send specialised commands to it. I use the "about" tab where you can send custome commands over the serial port. Here I noticed that even though my command gets to the arduino device fine, the hatire plugin reports a timeout. It isn't really a timeout, its is an effect I think of us having such busy serial communications in this scenario. I am looking at ways of getting rid of this issue, but to be honest, it is pretty minor so no big deal, everything still actually works.

I am definitely thinking about how to take gyro, accel and magnetometer data into the plugin. At the moment I use RTIMULib on my teensy to do full 9dof integration on the device. This is great, but I can only do this well because the teensy has good speed and memory. Some of the smaller arduino processors make this challenging with the smaller memory spaces. It will take me quite some time to look into this as I am quite busy with other things right now, but it's on my list of interesting things to do, especially since sensor integration is one of the big barriers for people trying to create their own trackers because of the vector/quaternion maths!

zakamik, 2015-04-25:
Just FYI - none of the above versions/plugins worked for me. Atmega328 + GY-85.
I guess some extra feature to see debug output would be nice to have.

[216 more replies]