opentrack/opentrack #140: FreePIE IMU issues when screen in standby

Mirkic7, 2015-03-11:
I'm using Samsung S3 mini with stock ROM.
All works fine while screen is on.
When screen goes off - centering for YAW gets very fast messed up.
However, if I just turn screen on, it immediately resets to proper values.
I've also made a vid of issue: http://youtu.be/vc5zgh7sZQg

Any ides how to fix this?

It's apparent some issue either with FreePIE IMU having some issues when phone is almost in standby, or even phone error in same mode.

sthalik, 2015-03-13:
There are some firmware-specific issues like wifi not working despite wifi lock but in this case looks like sensors return garbage. Screen can't be completely prevented from being turned off with a wake lock since the power button drops screen locks forcibly.

We're doing what can be done on Android to send data and not allow the device to sleep. The platform doesn't allow for more.

sthalik, 2015-03-20:
I've been able to reproduce it and there's a workaround available. It could take a while till it makes into FreePIE.

denilsonsa, 2015-03-20:
FreePIE probably already gets a wakelock anyway.

Still, have you tried using the SCREEN_DIM_WAKE_LOCK, by installing [this app][1]? Of course, this will not let the screen turn off, and will eat more battery, but maybe it works better. At least as a workaround until FreePIE gets updated.

[1]: https://play.google.com/store/apps/details?id=eu.thedarken.wl

sthalik, 2015-03-25:
In the meantime you can use this version <https://www.dropbox.com/s/kv4nrw2p3gw7ptz/app-release.apk> and confirm if it works. Just keep in mind it's signed with a different key and you have to uninstall the official one first.

davidcastellani, 2015-04-29:
I wanted to comment that I have had the same issue using Opentrack 2.3 rc9, with FreePIE IMU on a Nexus 4. I took the advice of @denilsonsa, and installed Wakelock. This has significantly reduced the "drift" I was experiencing. Its not perfect, but it seems Opentrack along with the sensors in my Nexus 4, are able to maintain "center" for much longer periods as long as the device screen stays awake.

Thank you so much for this software. It's made playing Elite: Dangerous, Star Citizen, etc. so much more immersive. I **_really_** enjoy it.

sthalik, 2015-04-29:
@davidcastellani you could check the version linked in my previous post. I'd like to get it included in next FreePIE version and independent confirmation would help.

cosmicdan, 2015-05-03:
Hi sthalik,

The test APK you shared has no effect for me (an engineering-board based on the MT6589 SoC) although my issue is different, the polling frequency drops drastically when in sleep. Checked my kernel sources and disabled power savings on sensors, albeit to no effect.

Since I'm a software engineer for eLinux/Android, I am comfortable with hacking around the FreePIE client app - maybe I can figure something out, even if only for those of us on rooted / engineering devices. Though I expect there will always be users that have to put up with setting their phone to infinite screen timeout. Will let you know if I get anywhere with my particular problem :)

sthalik, 2015-05-04:
Thanks guys! We're getting closer to release now.

closed 2015-05-04 by sthalik

vadirthedark, 2016-11-08:
I am using MDCCT and when I ./plot I get No such file or directory. What is the program looking for on the plotting drive?
Thanks

Mirkic7, 2016-11-08:
Which options did you use?

vadirthedark, 2016-11-08:
Using original algorithm (MDCCT). I have a 5TB drive to plot completely.The
cpu is AMD 4 core with 2MB L2 with 4 thdss and gpu is gtx 750ti with 4GB. I
wasn't sure - is this cpu plotting?

options used (1st plot A 885 n/m):
./plot -k ......................9587 -x 0 -d /media/imbota/burst/ -s 0 -n
19712400 -m 16384 -t 2
Adjusting total nonces to 19726336 to match stagger size
Creating plots for nonces 0 to 19726336 (5174 GB) using 4096 MB memory and
2 threads

Using this URL:
www.facebook.com/permalink.php?story_fbid=399055686908622&id=381160565364801
for instructions.

I need to adjust no's the GB for plots is too big. As you can see got it
working (slowly), any how to instuctions?
Thanks for any help you can give.

Mirkic7, 2016-11-09:
Hi,

my version of dcct plotter just adds SSE4/AVX2 optimizations to code and
async option for write (used double memory when using it with -a option).
Your syntax is OK.
However, if using 4-core CPU, use all 4 cores by setting '-t 4'.
Also, this version of plotter is CPU only. Please look for GPU version if
you want to use GPU aswell for plotting.

On Tue, Nov 8, 2016 at 11:12 PM, vadirthedark <notifications@github.com>
wrote:

> Using original algorithm (MDCCT). I have a 5TB drive to plot completely.The
> cpu is AMD 4 core with 2MB L2 with 4 thdss and gpu is gtx 750ti with 4GB. I
> wasn't sure - is this cpu plotting?
>
> options [...]

vadirthedark, 2016-11-09:
Thanks Mirkic7,

Let miner run about 20 hrs., this is readout:

8 Percent done. 1414 nonces/minute, 204:10 left

Does the 'left' number indicate hours? If so, I may go to GPU plotter.
I have tried 'gpuPlotGenerator', best I could get was 1024 n/min.
Thanks again for your help. Also, are you still mining and is your wallet
still available?

Mirkic7, 2016-11-09:
Yes, that means 204 hours left!
Yep, quite a long time :)

Still mining a bit BURST, but no need for donations :)

Enjoy!

vadirthedark, 2016-11-09:
I guess I will go to Plan B and try the gpu plotter - hope it will be
better.
Thanks again.

denilsonsa, 2016-11-09:
@Mirkic7 and @vadirthedark, why is this related to OpenTrack (a head-tracking software)? And why is this related to this issue that has been closed for more than one year? I'm talking about this: https://github.com/opentrack/opentrack/issues/140

vadirthedark, 2016-11-10:
Went to sse4 setting:

Creating plots for nonces 0 to 18907000 (4959 GB) using 12775 MB memory and
4 threads
4 Percent done. 2451 nonces/minute, 122:39 left
Also, what spec's. are involved in fast efficient gpu plotting and what
graphics card spec's are most important (or point me in the info direction)?

sthalik, 2016-11-10:
Wrong repo.