opentrack/opentrack #141: Where are all accela settings gone?

ruPaladin, 2015-03-17:
I wanted to try Opentack 2.3 but found that it has less accela settings than 2.2. Also I find microstutter of camera when use it. Please make it possible to change deadband, order 2 and 3, exponent.

sthalik, 2015-03-17:
Back in 2.2 there were too many settings. The way they interacted wasn't easily understood, beyond their obvious function. We're moving toward greatly simplified filter settings.

I don't know what "microstutter" means in this context. Please explain. Also provide your current settings. Maybe it's possible to make it work for you without further code changes, as multiple people were able to use the filter in its simplified shape.

sniperwolfpk5, 2015-04-26:
accela in 2.3 is not stable. There is micro stutter. Can any one bring accela from Facetracknoir. It is the  best which allow you your own settings.

asarnow, 2015-04-30:
I think they are referring to the spike noise - probably this issue should be merged with the 'jumpy' issue.

alemansec, 2015-04-30:
Having that issue with 2.3 too (unable to get steady values - semi-random spikes / jumps all the time).
Reverting to 2.2a2 in the meantime (i have to set some insane values there, to get steady positions, with acceptable responsiveness. That being: Rotation=200 Translation=200 Order2=80 Order3=150 RotDeadband=0.4 TransDeadband=0.4 Exponent=1.5).

asarnow, 2015-04-30:
@alemansec are you using a PS3 Eye? Can you see if current 2.3 works better if you lower the gain somewhat in CL-Eye Test? (CL-Eye menu > Options > Video Capture Filter).

sthalik, 2015-05-04:
The values are really extreme. Are your model settings accurate? You may also want to try hysteresis in PT tracker, it can reduce noise somewhat.

sthalik, 2015-05-17:
See: #149

closed 2015-07-12 by sthalik