opentrack/opentrack #144: Opentrack 2.3 release not working

Denco96, 2015-03-20:
Even though the old 2.2 release is working for me I can't get the new version to work with delan clip. When I press start I get my normal three points but another red point right of them. Checking the main tracker option it says extracted points: 3 OK. Main tracker is PoinTracker and the protocol is FreeTrack. I'm using the same settings as in 2.2 release so I see no reason it is not working.

Raw tracker data shows movement but it doesn't translate to game data no to the picture above.

forteh, 2015-03-22:
Have you calibrated the clip in the tracker settings?

Denco96, 2015-03-22:
I tried but the movement is not translating.

forteh, 2015-03-22:
Try creating a config from scratch, it's possible that a setting has translated badly from the earlier version. Also manually resetting and centering the tracker might help.

Denco96, 2015-03-22:
I've completely reinstalled it without any profile settings. What I meant with the above was that I was using the same settings in the main tracker than in previous version. Here is a picture of the issue I think is causing this. 

http://i1251.photobucket.com/albums/hh544/denco2/OpenTrack.png

Please note that if I try to run the previous version of OT 2.2 it works without issues.

Denco96, 2015-03-22:
Ha. I misslooked that the new version creates a Test folder and stores all the settings there. Its fixed now :)

forteh, 2015-03-22:
Do the tracked points move and alter the raw tracker data? It looks as though it is using the calibration from your previous settings.

The red cross denotes the centre of head movement (base of skull), with rc9 the auto calibration inverts the Y coordinates of this and displays it above the head.  What happens when you click the auto calibration button?

Denco96, 2015-03-22:
I cant find the auto calibration option but I did managed to fix my issue with deleting the test folder.

sthalik, 2015-03-25:
What test folder? It shouldn't be happening. We're using last directory a config was saved to.

closed 2015-03-31 by sthalik