opentrack/opentrack #145: Range of pitchaxis

IvanDolvich, 2015-03-26:
Hi
In version 2.3 rc9 I am having problems with the pitchaxis. The game I am using is IL2 Battle of Stalingrad. When I'm trying to look up the maximum is around 45 degrees (pitching up). 
In 2.2 stable this is not an issue. There I can pitch up around 90 degrees.

forteh, 2015-03-26:
Is the tracker still able to track all points when fully pitched up? I believe that if the tracker loses a point it will freeze at that position and restart when it regains the point.

If the tracker registers all points through the full range of motion, have you set the mapping range to full? Are there axis curves that you can control within the sim that might need expanding to suit the output from rc9?

IvanDolvich, 2015-03-26:
LEDs are all still visible and the curves are alright and downwards 90 degrees is possible but thanks.

forteh, 2015-03-26:
Odd, are you using assymetric axis mapping for pitch? When you pitch up and down does the raw tracker data read the same? Mine pitches up and down 30° equally.

This is my pitch mapping (primarily for arma3 where I want to be able to look up more than down). Nominally the vertical axis is the range of movement in game, the horizontal axis is the range of movement of your head.  Apologies if you already know this and I am explaining for no reason :)

[image]

Have you centred the tracker after starting opentrack?

IvanDolvich, 2015-03-27:
Please do not apologize for helping people :)
BTW thanks for your picture forteh. 
I am using asymmetric mappings for the axis i have also checked the raw input-data. 
With my problem I think centering doesn't help to solve.
Some time ago in the IL2 forums I think I remember a thread that mentioned the same problem as I am having now. One guy there recommended to switch to the "old" 2.2. In this version the 90 degrees pitchup does work

forteh, 2015-03-27:
When pitching does the game data value rise up until the peak value on the vertical mapping axis?
For example at approximately +/- 30° pitch my game data values are +120 / - 95 respectively; the tracking point should move up/down the curve (when the tracker is active) to reflect this.

If your tracking point isn't going above 90° on the vertical pitch axis but the curve continues then perhaps you need to drag the final curve points to the left a little to steepen the mapping curve.

If your tacking point is going over the 90° point on the vertical pitch axis but it's not translating into IL2 then I'm afraid I'm stumped and the only thing I could suggest would be to try to alter the view axis curves in the sim (not even sure if this is possible but I think you can do it in dcs a10c).

IvanDolvich, 2015-03-27:
I see what you mean here. Tanks a lot for your effort.

Len62, 2015-03-30:
Works okay for me in BOS. Make sure on the pitch curve you set the upper limit at 180 deg, not 90.

IvanDolvich, 2015-03-30:
Sound like a quite easy solution. And the best is, it works ;)
It didn't come to my mind that you need to set the pitch-curve up to 180 degrees to get the ingame-result of 90 degrees. Thanks a lot!

Len62, 2015-03-30:
You're welcome! I don't remember where I got the info from, may have been here, and it is unusual to set 180 to get 90. Oh well, glad it worked. Happy Flying!

sthalik, 2015-04-04:
Glad you guys figured it out!

closed 2015-04-04 by sthalik