opentrack/opentrack #148: Mixed tracker aruco/udp for Gagagu VR Streamer

gagagu, 2015-04-01:
Hi,
i've written an small virtual reality iOS App for play PC games on a VR Headset like google cardboard on iPhone (more info http://www.gagagu.de).
I'm using the Opentrack with UDP Tracker for Headtracking and all is working really fine. But with my iPhone i only abe to send yaw, pitch, roll data to the udp tracker (x,y,z is always 0.0) because the phone do not have any sensors to measure positional tracking.
Now i'm thinking about how to realize the positional tracking (x,y,z) and i'm wanted to use the aruco tracker for it. Now my question: is it possible to create a new tracker which is a mix of udp and aruco tracker? It means yaw/pitch/roll from udp and x,y,z from aruco? I have a little bit c++ knowledge, is it possible for me to compile my own tracker dll, and how?

I'll be sure you want to ask why i'm not using aruco for yaw,pitch,roll? Because the sensor data from iPhone are faster and more precise.

If you like to contact me for further explanations here my mail address: gagagu@web.de

Thanks forward!

sthalik, 2015-04-01:
Hey,

To build the software, a modern C++11 compiler is needed. Ones known to work are Clang and GNU. You can grab a full compiler/runtime suite for Windows from the mingw-w64 project <<https://sourceforge.net/projects/mingw-w64/files/latest/download>>. Additionally CMake is used to generate the makefiles. Visual Studio is known not to work, as it only supports few features from C++11.

Qt5 is a required dependency. There are several optional ones, like OpenCV to build Aruco and several other trackers depending on OpenCV functionality.

You can skip building Aruco and even the executable if you use the exact Qt and compiler runtime as the official executable for 2.3 release candidate:

- GCC 4.9.2
- mingw-w64 exception protocol **dw2**
- Qt 5.4.1

Keep in mind Qt5 has to be built by the same compiler for the ABIs to match.

Then, the .dll generated will be compatible with official opentrack builds.

I understand this is complex for Windows users. It's way easier to do on OSX or Linux.

To create an actual tracker, include "**opentrack/plugin-api.hpp**" and use the **ITracker** and **ITrackerDialog** interfaces from "**opentrack/plugin-qt-api.hpp**".

Inside your own tracker code, link at runtime to both of the UDP and aruco trackers. Invoke the **GetConstructor** function exported by the library:

<code>extern "C" OPENTRACK_EXPORT ITracker* GetConstructor()</code>

On the newly-created tracker object call **start_tracker**:

<code>virtual void start_tracker(QFrame* frame) = 0;</code>

Note QFrame pointer is necessary for tracker feed display, but an empty and invisible frame can be used just the same.

Then, calling **data** will provide values from the tracker:

<code>virtual void data(double *data) = 0;</code>

Then, you can use the values provided in a newly-created tracker.

gagagu, 2015-04-01:
Uff, thats hard... firstable thank you for your reply. I don't know i will get it. I will first start to compile the udp tracker and see if its working.

gagagu, 2015-04-01:
Hi, where do i find the ftnoir_tracker_udp.h ? thx

gagagu, 2015-04-01:
Hello again, i've made a small template for my plugin and want to compile and link it. The compilation is working without aný issues but the linker is given some Errors. I think he is missing some libs but i do not know which. I've made a small zip with my files for better understanding.
http://www.gagagu.de/ftnoir_tracker_gagagu.zip
If possible it would be great to get some Little help. Thanks Forward.

Error Message:
C:\tmp\Tracker\ftnoir_tracker_gagagu>g++ -c -std=c++11 -I C:\tmp\Tracker\opentra
ck-opentrack-2.3-rc9 -I C:\Qt\Qt5.4.1\5.4\mingw491_32\include\QtCore -I C:\Qt\Qt
5.4.1\5.4\mingw491_32\include\ -I C:\Qt\Qt5.4.1\5.4\mingw491_32\include\QtWidget
s -I C:\Qt\Qt5.4.1\5.4\mingw491_32\include\QtGui ftnoir_tracker_gagagu_dll.cpp

C:\tmp\Tracker\ftnoir_tracker_gagagu>g++ -shared -o ftnoir_tracker_gagagu_dll.dl
l ftnoir_tracker_gagagu_dll.o C:\Qt\Qt5.4.1\5.4\mingw491_32\lib\libQt5Core.a C:\
Qt\Qt5.4.1\5.4\mingw491_32\lib\libQt5Widgets.a C:\Qt\Qt5.4.1\5.4\mingw491_32\lib
\libQt5Gui.a
ftnoir_tracker_gagagu_dll.o:ftnoir_tracker_gagagu_dll.cpp:(.text$_ZN7QStringC1EP
Kc[_ZN7QStringC1EPKc]+0x36): undefined reference to `__imp__ZN7QString16fromAsci
i_helperEPKci'
ftnoir_tracker_gagagu_dll.o:ftnoir_tracker_gagagu_dll.cpp:(.text$_ZN17FTNoIR_Tra
ckerDll4iconEv[_ZN17FTNoIR_TrackerDll4iconEv]+0x37): undefined reference to `__i
mp__ZN5QIconC1ERK7QString'
ftnoir_tracker_gagagu_dll.o:ftnoir_tracker_gagagu_dll.cpp:(.text$_ZN15QTypedArra
yDataItE10deallocateEP10QArrayData[_ZN15QTypedArrayDataItE10deallocateEP10QArray
Data]+0x1e): undefined reference to `__imp__ZN10QArrayData10deallocateEPS_yy'
collect2.exe: error: ld returned 1 exit status

gagagu, 2015-04-02:
Hi, thx but i could not get it to work. Its too complicated for me to compile. Too many options with too many error sources i don't understand. Normally i'm using Visual Studio.

sthalik, 2015-04-04:
There's nothing that can be done about it at the moment.

closed 2015-04-04 by sthalik