opentrack/opentrack #149: Jumpy on X and Y axis

Trooph, 2015-04-17:
Hello Stanislaw. Its been a long time since i posted here.
Maybe its not an issue but its worth pointing this.
After tweaking my accela settings and stuff i managed to do awesome settings. Smooth and accurate as Track Ir. But there is this things that annoys me.
When not moving my head at all tracking keeps going down on Y axis and sometimes to -X axis aswell its a jump of few pixels but its noticable. Its possible to post a yt vid. 
It happens in all sim games i have. Even arcade thunder 

T.

forteh, 2015-04-20:
Have you got a small deadzone set at the start of each map curve?  To be honest I don't think I can hold my head perfectly still so as to not give an input to opentrack, I haven't set it up on a fixed point but I'm pretty sure it's rock solid for me.

asarnow, 2015-04-30:
I'm having a lot of spiking in 2.3, even with "extreme" filtering. The deadzones take care of the noise when the view is centered only, there is a lot of jumpiness otherwise.

sthalik, 2015-05-04:
With translation sensitivity slider to the right you should be able to smooth it out more.

I'll think of a separate smoothing slider for translation but only if this issue can't be solved otherwise.

Trooph, 2015-05-05:
Thanks for the response 
@sthalik Actually my sliders are all the way to the right. That strange jumping happens even on extreme filtering..

asarnow, 2015-05-05:
@Trooph Is the jumping in the raw values? What I am seeing is jumping or oscillation in the raw values of all axes; the centers of the LED points hop 1 or 2 pixels from frame to frame. Going up to 'extreme' spike noise reduction helps, but it's not enough. Using 640x480 resolution ameliorates the problem quite a bit, but restricts the PS3Eye to lower frame rate.

In 2.2 this still happens, although apparently less so, and in that version of Accela I have the 2nd and 2rd frame orders very high (2.0 and 3.0 respectively) and with a large deadband so the filtered output is OK.

sniperwolfpk5, 2015-05-06:
Yes i confirm that there is jumpiness. Like some one kick me. You can't click due to jumping. Is it possible to use facetracknoir accela. As i use facetrack because of it's stability. There are option that i like in opentrack but due to jumpiness it is useless. 
By the way i am using version 2.3 rc9

Trooph, 2015-05-06:
Ill upload a vid tomorrow. It happens when im siting completely still. Its only X Y Z axis that suffer from that and sometimes it does make clicking those buttons harder :)

FlyingCircus-, 2015-05-08:
Might it be the lack of filtering deadzones that I pointed out some time ago? I did notice how difficult was to get a completelly steady pose, usually due to very small variations in the raw coordinates (even below the single digit degree we see as feedback) that were greatly multiplied to several degrees of drift due to the mapping curves. I think we would all benefit of having filtering deadzones (before the mapping curves are applied).

FlyingCircus-, 2015-05-08:
To be more clear: with deadzones I was referring to the adjustment of the sensitivity so, for example, only 0.5 or 1 or whatever raw degrees variations are taken into account.

asarnow, 2015-05-09:
@FlyingCircus- Unless I'm missing something, in 2.3-rc9 we have only rotation / translation smoothing amount and several presets using EWMA filter (if I read the code correctly) for 'spiking noise reduction' - similar to past versions 2nd/3rd frame weights or filter deadband in effect.

FlyingCircus-, 2015-05-09:
@asarnow It's my understanding we actually lost real deadbands in Accela since we got the merged settings. Noise reduction is done by averaging values in the "game values", but I believe we need a way to completely discard small movements of the "raw values". At least, that was my interpretation of what was happening.

sniperwolfpk5, 2015-05-10:
Facetracknoir accela  is rock steady. Why not use facetrack accela.

asarnow, 2015-05-10:
@sniperwolfpk5 Accela Mk. 2 requires manually drawing the filter curves. @FlyingCircus- You're correct that there is no true deadband in the filter response. IMHO it's hard to understand the filter from the code alone, so I made [some visualizations] (https://imgur.com/a/BZO7N) comparing the current and Mk. 4 Accela versions. It looks like the current filter is better than Mk. 4 in a high S/N regime, while the aggressive smoothing and deadband options in Mk. 4 are better in case of low S/N. Finally, the 'high RC' images suggest that we may be able to solve our problems with higher RC values.

FlyingCircus-, 2015-05-14:
@sniperwolfpk5 Those are very cool visualizations, Did you extract the filtering functions in order to plot them? Anyway the current filter is perfectly usable for me even if I think it could improve with customizable deadbands. I have gain set to its lowest while keeping very high exposure in the PS3eye driver, you might test it with those settings.

sthalik, 2015-05-14:
@asarnow awesome plots!

I'll introduce deadbands for rotation and translation for the next version, just give it time.

sniperwolfpk5, 2015-05-15:
@FlyingCircus this is not done by me. These visualization i done by @asarnow.  @sthalik It would really nice to see dead bands.

asarnow, 2015-05-17:
@sthalik @FlyingCircus- Thanks, all I did was port the filter code to a MATLAB/Octave script. I'll commit the script into my opentrack fork, if you're interested in the details. I've also got a working build environment for opentrack now (plus a how-to in my fork's wiki), I look forward to testing out the new changes. I know it's possible to have a simple UI *and* good tracking even with low S/N. I designed my IR clip to be safe (under OSHA limits for IR exposure) even with unlimited direct viewing from < 5 cm. Definitely a trade off there, but it does work very well with appropriate filtering.

dbaarda, 2015-05-17:
Very interested in the working build env wiki. I can probably contribute
lots on what *doesn't* work :-)

asarnow, 2015-05-17:
@dbaarda  [Here] (https://github.com/asarnow/opentrack/wiki/Build-Instructions) is the page.

sthalik, 2015-05-19:
Please see if the functionality works properly in rc10. I've only lightly tested it.

FlyingCircus-, 2015-05-19:
@sthalik The translation deadband definetively helps me to steady the output, no doubt about that. Regarding the rotation deadband I would like to have a much higher deadband, at it's current top value (0.3º) the difference is negligible. I would like to see it reach 1º or even 2º, as I said some time ago, I sit very close to the camera and even the slighlest movements are in the order of several raw degrees.

sthalik, 2015-05-22:
Okay, it's gonna be two degrees for rotation.

sthalik, 2015-06-02:
Is it good in rc12? Can we consider accela mk5 suitable replacement for mk4 and below?

sthalik, 2015-06-15:
@dbaarda would it help you if opentrack included code for automatic checkout and build of all its dependencies? This still requires a working compiler. We technically could ship premade GCC for x86 with the codebase. Is this what causes your issue?

sthalik, 2015-06-16:
@FlyingCircus- @Trooph @forteh 

I'm testing a new simpler method for extracting points from camera image. Please tell me if this build makes it worse requiring more filtering or is it fine. It can be further improved but first we need to know if it works at least as well as the old method.

If you get centering/mapping problems please also tell me, there are also changes.

https://www.dropbox.com/s/3lnx18rds2jtyn6/opentrack-test-pt-extraction-20150616.7z

asarnow, 2015-06-18:
Haven't been able to test much due to thesis, but:
* RC12 didn't work well for me. Neither did custom builds with higher deadzone, etc. values.
* I couldn't figure out the exact mechanism despite lots of theoretical investigation (basically because of how smoothing and curves/dz were not as separated in Mk.4).
* I had a little time today, new test build seems *much* better, I get good behavior with less aggressive filtering (all parameters). Maybe the culprit was always point extraction.
* I seemed to have a little trouble with optimal camera thresholds, but there was a lot of ambient sunlight and I didn't have time to explore all camera gain levels. Probably it is fine.
* I've been meaning to take a look at some automatic threshold approaches as I have some expertise in that area and several approaches are fast enough (e.g. EM algorithm with Poisson distribution).

@sthalik @dbaarda About building, MXE gives pretty good automatic dependency handling, only downside right now is the higher number of DLLs and the CMAKE little cmake bug described in my fork's build instruction. I think those can be fixed pretty easily, especially the latter.

sthalik, 2015-06-19:
@asarnow if indeed the new point extraction is better performant, it can be further improved by taking additional binarized frames with thresholds close to the user-specified threshold, and averaging point's location basing on how circular it is. This is OpenCV's approach.

Could you please provide a video of keeping still with model coordinates, so that results of these multiple approaches (old one, test one, test one with multiple thresholds) can be compared to keep the most idempotent one?

sthalik, 2015-06-19:
@asarnow what was wrong with new accela sensitivity code? Was it jumpy or too slow to react?

sthalik, 2015-06-19:
I need a camera feed of someone's model being idle and stationary. Bigger the resolution and point light size, the better. This is to test point extraction accuracy with the new code.

sthalik, 2015-06-19:
This camera feed video best to leave the model on a desk somewhere and record for few minutes, then record. I also need the model dimensions. Please compress the video so you can send it with no problems.

sthalik, 2015-07-10:
@dbaarda you can build using Cygwin's "mingw64" toolchain, using "ccmake" from Cygwin to configure. The toolchain is prebuilt.

closed 2015-07-16 by sthalik