opentrack/opentrack #151: New Plugin Template

gagagu, 2015-04-27:
Hi,
i want to create a new Plugin for Opentrack. It should be a mixed Plugin between  aruco and UDP sender (There is aready an issue but its closed and i don't know how to reopen; https://github.com/opentrack/opentrack/issues/148). 
I've created a small template to create a plugin (https://github.com/gagagu/ftnoir_tracker_gvrs). I get no errors from compiler or linker. When i will use the generated dll to opentrack the tracker will not shown in list. I have no idea whats wrong?
I've comment out the "ui_ftnoir_ftnclientcontrols.h" because i cannot find it (it think it will generated by facetracknoir). May be thats the reason?

Which license type i should use for a opentrack plugin project on github (i'm new on this)?

Thanks forward.

sthalik, 2015-04-27:
The ui header is generated by Qt "moc" during compilation.

In the .dll in your repo there are no "GetConstructor" and "GetDialog" symbols, only "GetMetadata" exists. Open it up with Dependency Walker.

The UI dialog is optional, return NULL in GetDialog. Given you can use a different Qt version, you'll run into crashes otherwise.

gagagu, 2015-04-27:
Hi, thx for reply.
A fisrt big step is done. I found out how to build opencv and opentrack. I only got some linking errors but i'm working on. I's very hard to find out which lib files are have to build in.

My project is a copy from ftnoir_tracker_udp without receiving udp data (it only sends "0" for test). The GetConstructor is in ftnoir_tracker_gvrs.cpp and the GetDialog is in ftnoir_tracker_gvrs_dialog.cpp.

Greetings

sthalik, 2015-04-27:
You can attach your code to opentrack build process. Then you won't have to mess around with manually compiling and linking. The exported symbols listed are not in the .dll in the repo. They exist in working builds.

gagagu, 2015-04-27:
Hi, thanks for the great tipp. I am able to build the plugin template.
Now i'm working on theplugin code. I've loaded the dll and got the Constructor.

	QString fullPath = QCoreApplication::applicationDirPath() + "/" + "libopentrack-tracker-udp.dll";
	handle = new QLibrary(fullPath);
	Constructor = (CTOR_FUNPTR) handle->resolve("GetConstructor");

Now i have to think about to go further :-)
thx

gagagu, 2015-04-28:
Hi,
i've inserted the udp tracker code into my template and everything is working good.
Now i want to Access the aruco tracker dll by following code
typedef void* (*TRACKER_PTR)(void);

in constructor:
QString fullPath = QCoreApplication::applicationDirPath() + "/" + "libopentrack-tracker-udp.dll"; 
handle = new QLibrary(fullPath); 
if(handle){ 
 	ptrTracker = (TRACKER_PTR) handle->resolve("GetConstructor"); 
 		//if(ptrTracker)  // comment out to  cause a Crash when empty, for test
 			ptrTracker(); 
 	} 

This seems to run withut any error but run() and start_tracker() are not working. The Programm will Crash.

code in run():
 	if(handle){ 
		ptrTracker = (TRACKER_PTR) handle->resolve("run"); 
		ptrTracker(); 
 	} 

typedef void* (*TRACKER_PTRSTART)(QFrame*);

code in start_tracker():

if(handle){ 
	ptrTrackerStart = (TRACKER_PTRSTART) handle->resolve("start_tracker"); 
	ptrTrackerStart(videoframe); 
  }

what i'm doing wrong here? And how do i Access the data from the aruco tracker in the data method?
If someone interested to see the code: https://github.com/gagagu/ftnoir_tracker_gvrs

Thanks Forward for helping me!

gagagu, 2015-04-29:
Hi,
 i've inserted the udp tracker code into my template and everything is working good.
 Now i want to Access the aruco tracker dll by following code

typedef ITracker* (*TRACKER_PTR)(void);

QString fullPath = QCoreApplication::applicationDirPath() + "/" + "libopentrack-tracker-udp.dll";
 handle = new QLibrary(fullPath);
 ptrTracker = (TRACKER_PTR) handle->resolve("GetConstructor");

ptrTracker->run();

But the last line doesn't work. I think the reason ist that the typedef should be:
 typedef Tracker* (*TRACKER_PTR)(void);

But then i have to include the ftnoir_tracker_aruco.h and it cannot find the "ui_aruco_trackercontrols.h"
 Where do i find it?

Thx

gagagu, 2015-04-29:
Hi, got it. I have to insert
link_with_dinput8(opentrack-tracker-aruco) to the CMakeLists.txt

gagagu, 2015-04-30:
Hi,
thanks for your help. I've got my plugin running, now. I've insert the code for apple too, but i cannot test it. I want to release my code and compile dll on my github project. 
https://github.com/gagagu/ftnoir_tracker_gvrs

Do i have something to consider in view of license?

Thx

sthalik, 2015-04-30:
There are several issues here for Unix. No dlclose call in destructor. Extension is .so, not .dll.

The main executable still contains GPL3 bits. It's best to use ISC license since it's so short: https://github.com/opentrack/opentrack/blob/unstable/ftnoir_filter_accela/ftnoir_filter_accela.cpp#L3

gagagu, 2015-04-30:
Hi,
i will update this issues as soon as possible. thx for helping me

gagagu, 2015-05-05:
Hi, i've updated the code for other operating systems than windows. I hope it works.

sthalik, 2015-05-05:
It's done on a separate thread and at reduced speed. It takes 16 ms to render frames at 60 fps so it's very hard to be late with pose data.

sthalik, 2015-06-01:
I don't like how it's necessary to duplicate code from plugin-support.cpp to get it to work. opentrack should provide a header with all the logic necessary to open libraries on all platforms.

Let's leave this issue open until there's a fix.

sthalik, 2015-06-07:
@gagagu you can now include "opentrack/plugin-support.hpp" to gain access to opentrack plugins. It abstracts away OS differences in opening shared libraries.

I recommend using the Modules class, trackers() and then grabbing a tracker with your preferred tracker->name.

gagagu, 2015-06-07:
Hi, Tod für your help. I am very busy atm. I will check it in a couple of days.
Greetings

gagagu, 2015-06-16:
Hi, 
i have insert the 
#include "opentrack/state.hpp" to my ftnoir_tracker_gvrs.h and defines in private:
Modules modules;

I want to use modules.trackers() with a for loop later in my constructor method.
When i will make it the linker will give the error:  undefined reference to 'dylib::enum_libraries()' 

When i will comment it out the building will run without errors. 
A little help for a beginner would be great. Thanks forward.

sthalik, 2015-06-18:
@gagagu state.hpp depends on selected-libraries.cpp so it's not safe to include on its own. Why do you need it? plugin-support.hpp should be enough.

sthalik, 2015-06-18:
Try this:
<code>
Modules m;
auto list = m.tracker_modules;</code>

Then you can iterate over the list looking for aruco.

gagagu, 2015-06-18:
Hi, i need the state.h because the class or struct  Modules is not known by compiler.
error: 'Modules' was not declared in this scope
Modules m;

By the way: the file #include "opentrack/plugin-support.hpp" does not exists. there are only a plugin-support.h do you mean this header?

sthalik, 2015-06-18:
No, you're probably using an old build.

gagagu, 2015-06-18:
I'm using 2.3 rc12 p1

gagagu, 2015-06-19:
Hi again,
i costs me a lot of time to get the actual version compiled (problems with opencv) but at the end i got it.
I've updated my code and implement the stuff you told me. If you like you can take a look to my (now short) code here:

sthalik, 2015-06-19:
Your destructor is OK. mem<t> is std::shared_ptr<t> and setting it to nullptr frees the memory.

Congrats on getting it done!

gagagu, 2015-07-06:
Hi, i've build my plugin with the new released source rc13 but my plugin is now not visible in the trackerlist anymore. I cannot find the difference what i have to change. On rc12 and the unstable build (17 days before) the plugin was visible and working. Thx forward for help.

sthalik, 2015-07-06:
Send me the broken dll.

sthalik, 2015-07-06:
And don't do this:

    #include "opentrack/selected-libraries.cpp"

If you have to it's a bug in opentrack. But you shouldn't have to.

gagagu, 2015-07-06:
Hi, i've send you the dll and the code. Thx

gagagu, 2015-07-06:
I've changed the Mingw32 to sjlj and now the dll is working with 2.3 rc13
Thank you for the special code change!

sthalik, 2015-08-21:
@gagagu there are now comments in plugin-support.hpp. You should get along fine with the changes made. Closing this issue.

closed 2015-08-21 by sthalik