opentrack/opentrack #152: HT doesn't work in 2.3RC9

njc, 2015-04-30:
I'm using an unmodified PS3 Eye and face tracking doesn't work for me.
The device is configured correctly (640x480@60fps) and I get a feed from the camera in Opentrack, but no movement is registered. Using FreeTrack protocol and Accela filter.

sthalik, 2015-05-04:
Are you wearing glasses or looking at a weird angle? Can you submit an image?

njc, 2015-05-04:
Neither. The webcam image shows up alright, but no tracking is performed
(no raw or game coordinates). Are there any log files I should include or
is it just a screenshot?

njc, 2015-05-05:
Here's a screenshot. Webcam seems to be connected ok (works with
facetracknoir).

man. 4. maj 2015 kl. 18.57 skrev Nils Jakob Clemmensen <
nj.clemmensen@gmail.com>:

sthalik, 2015-05-22:
There's no screenshot. Attaching via main doesn't work, please use github feature.

njc, 2015-05-22:
Oh, here you go.
[image]

sthalik, 2015-05-31:
@njc cropping your face off the image and putting it through a virtual webcam source, it detects your face just fine. I'd like to give it further debugging but for now I can't reproduce the problem.

sthalik, 2015-07-16:
Response timeout

closed 2015-07-16 by sthalik