opentrack/opentrack #153: [Suggestion] Better mouse emulation?

Sokonomi, 2015-05-03:
I have found that many games do not pick up on the mouse emulation produced by opentrack. Though often the cursor works fine in the games menus, the camera does not move.

A program named XPadder seems to do a good job translating joystick to mouse, so good mouse emulation is definitely not inhibited by the games themselves. Unfortunately I am not savvy enough to figure this out other than "its not working"..

But since any game that doesn't allow TrackIR/joystick assigning for its camera control is reliant on the mouse, I think it deserves to have a look at improving the mouse emulation.

Games ive tried are GTA4 and 5, Farming simulator 2015, and a few RC flight sims.

sthalik, 2015-05-03:
Hey,

It's believed fixed for the next version.

closed 2015-05-03 by sthalik

Sokonomi, 2015-05-03:
When will the next version release though?
Its been three months since the last binary.

Nakonzaid, 2015-05-14:
First of all, sorry for my poor English.
I pray for that new version! :)
I need to use mouse emulation in  several games.
So, please, can you tell us a approximate date for that release?
Best regards and a big thank you!

Sokonomi, 2015-05-14:
Im also highly anticipating this improvement. It would make so many games head-track compatible.

Nakonzaid, 2015-05-14:
Yes! 
And very usefull for vr too