opentrack/opentrack #156: [Feature request] Support single point devices

MathijsG, 2015-05-17:
In FreeTrack you can use a single point with the tracker, to have a 2 DoF system with X and Y value.

Unfortunately OpenTrack doesn't support a single point system, this would be a warm welcome, to support basic headtracking with just a single led. Many games just need "looking around in the cockpit" which just needs one dot.

Is it a possibility to add a single dot feature in the pointtracker plugin?

sthalik, 2015-05-17:
> Is it a possibility to add a single dot feature in the pointtracker plugin?

Pointtracker isn't extensible for additional models.

If you can make a single dot model, you can just as well make a three-dot model for full 6DOF. I don't think it's necessary to support that.

MathijsG, 2015-05-17:
Many people aren't that technical to make a full clip or something, while with one dot model even a simple bicycle light ducttaped to a hat would be sufficient. I don't know how hard it is to add this, but I think it will gain a few users who do have a simple light and just want 2DoF.

sthalik, 2015-05-19:
With 2DOF we'd only get X and Y translation however. Rotation won't work reliably. Is it worth advertising support for something as poorly working?

MathijsG, 2015-05-19:
It's your program so it's your decision, but I wouldn't call 2DoF with X and Y poorly working. Those axes are working fine. 99,9% of the games just need something to look around with in a cockpit, who uses rotation anyway :).

FlyingCircus-, 2015-05-20:
@MathijsG I know it's not what you're asking for, but you might try the aruco tracker in case you wan't a low technology alternative. You just need to print the pattern and fix it somewhere visible from the webcam field of view... stick it in your forehead and you will be even crazier looking than the rest of us with 3 LEDs models ;)

MathijsG, 2015-06-01:
I already have some 3 point devices, the feature wasn't for me, but just for people like my friends, who find the whole soldering a clip a too hard thing to do. Here some video's with a single led, to see how well this can be working:

https://www.youtube.com/watch?v=UEx-24GRHkA

https://www.youtube.com/watch?v=WExFyTt4eJU

I think this will be a big usage boost for this software.

sthalik, 2015-06-01:
Can't they use the aruco tracker? It's just a b&w piece of paper.

It's not exactly easy to add a new point model in PT. I won't reject it outright, but it has low priority.

MathijsG, 2015-06-01:
I will check this allternative out myself. I understand if it is a hard job.

MathijsG, 2015-06-16:
Apart from the difficult job it seems to be to implement single point devices, the Aruco thing is in no way an alternative to me, maybe it's my webcam (the PS3 eye I made is only able to be used for infrared, so I had to use a regular webcam for this) but it loses the aruco all the time.

sthalik, 2015-06-16:
Is it a 640x480 webcam? It should alternate "box size" in latest version linked in another thread. Then it settles on the "box size" that works for given light and noise amount.  If you make a short video of camera feed I can look further.

MathijsG, 2015-06-16:
My webcam isn't ideal, so maybe it's not really representive. I will try anyways. I also have a spare PS3 Eye cam which I didn't kill for IR tracking. I will try that one.

sthalik, 2015-07-16:
I'll reconsider if someone decides to write a single-point tracker.

closed 2015-07-16 by sthalik

Luc45, 2015-08-09:
I use single-point tracker because I think 6DOF is distracting, as I only need to look around, and one LED uses less battery, wich is good because the batteries in my cap is annoying to change, luckly I never had to do it, but with 3 leds I think I would already.

MathijsG, 2015-08-09:
And what do you use for tracking at the moment? FreeTrack?

Luc45, 2015-08-09:
Yes, I use Freetrack, but there are some games that only runs with OpenTrack, like Project Cars for example

WillMazeo, 2016-01-28:
lol was trying so hard to get a single LED working and use this on Stock Car Extreme, now I found this thread. Frustrating :( 
Great program tho

MathijsG, 2016-01-28:
You can still use freetrack for this. They provide support for single led solutions.

WillMazeo, 2016-01-28:
Unfortunately that one is not working in Stock Car Extreme =/