opentrack/opentrack #157: Mouse Emulation still not functioning properly.

Sokonomi, 2015-05-19:
I noticed in the new release candidate that work has been done to mouse emulation.

Ive tested it in GTA5 and Farm Simulator 15, both still do not pick up the mouse emulation from opentrack while in game (oddly enough it does move the cursor when you open the in-game menu.

So the issue raised in my previous ticket unfortinately remains; https://github.com/opentrack/opentrack/issues/153

Nakonzaid, 2015-05-20:
I'll try in Skyrim, GTA-IV and others today. Then i'll post the results.
Anyway, thank you very much for the hard work!

Sokonomi, 2015-05-20:
I'll attempt a few more games as well.

I also have a small suggestion, dont know if its possible; A settable deadband zone where mouse emulation "lets go" so to speak. This has two major benefits; One is that the mouse wont keep creeping around after you toggle it off, the other would be the ability to navigate menus when you face the screen directly. The last part would help games that have of dialog menus where the mouse steals focus when you move it. (Fallout and skyrim)

sthalik, 2015-05-20:
It doesn't work currently. I have some idea on how to try to fix it again. Stay tuned.

Nakonzaid, 2015-05-21:
Thanks, sthalik!

sthalik, 2015-06-01:
The feature's been implemented for next RC. Please provide feedback.

MathijsG, 2015-06-01:
I tried in GTA V (with the previous build, RC10) and it is working perfectly. You need to change the settings in GTA V, the mouse settings should be on Windows instead of Raw. When on raw, the input is directly going to the game, instead of via Windows.

HornetMaX, 2015-06-01:
Works in-game for me too ! Nice job !

closed 2015-06-01 by sthalik

Nakonzaid, 2015-06-02:
Thank you very much!