opentrack/opentrack #158: Octopus turns around like crazy

thelaoshi, 2015-05-20:
Just tried opentrack (latest rc) + freepie on a sony z1 and samsung galaxy s2 and I have with both phones the problem, that  even if i dont move the phone the octopus moves like crazy as soon as i start freePIE.
Smallest movemens make the GameData values look like Randomized Number changing like will.
In Debug mode with PIE I see Orientation changes quite well and even if they dont change opentrack seems to detect movement?

I have no idea where the problem could be :-(

thelaoshi, 2015-05-20:
forget it. there are 2 apk's and using the one under android-freepie was the failure :)

closed 2015-05-20 by thelaoshi

sthalik, 2015-05-20:
My bad about 2 apk's! Will be fixed for the next update.