opentrack/opentrack #160: [Feature Request] Automatic profile loading (game.exe recognition)

Sokonomi, 2015-05-20:
I would like to see opentrack automatically loading a profile and starting when it detects a game associated with it, so you can just leave it running in the systemtray. Preferably making it go dormant again once no games are detected.

An example of this mechanic is the logitech gaming software, which lets you appoint a game.exe to a profile.

fawick, 2015-05-20:
FYI: PR #161 is for fixing #159, not this issue. Sorry.

sthalik, 2015-05-20:
You want to associate profiles with running executables? That makes sense, even though the current profile system might need a bit of an overhaul.

I'll try to get this change in before 2.3 since it's a cool feature to have.

sthalik, 2015-06-01:
The feature's been implemented for the next RC. Please provide feedback on what's missing, what to change, etc.

MathijsG, 2015-06-01:
I do like the feature, however while in-game it is sometimes randomly stopping with tracking for a few seconds, and then starting again. And when I alt-tabbed out of the game and manually clicked stop (to alter some settings) it wouldn't let me stop it and keeped starting it again (because the game is still running).

Maybe it would be handy that if you manually press the stop button it will override the autostarting while the game is running?

sthalik, 2015-06-01:
I agree about the stop button.

Can you tell me more why it randomly stops?

MathijsG, 2015-06-01:
No, it just randomly stopped tracking, I will investigate this further. I have a dual screen setup, maybe I can put up OpenTrack on my second screen and monitor both :).

sthalik, 2015-06-01:
I don't have a good idea on what to do when user presses 'stop'. A game is running, possibly several (!). User expects game detection to still work at all. There's nothing good to do then.

MathijsG, 2015-06-01:
But if a user manually alt-tabs to OpenTrack in my opinion you may assume he wants to edit something in OpenTrack, so when he manually stops the tracking it should stay off until he starts it again and alt-tabs back into the game? That would be my use case.

sthalik, 2015-06-01:
Alt-tabs can't be detected. It can be disabled until tracking starts manually again but is there no better solution?

HornetMaX, 2015-06-01:
Maybe have the "Auto start" checkbox just beside the start/stop buttons, so that it is clear that if you press stop and auto start is on, then it will re-start if game is executing.

Not sure what happens if running two games associated with two different profiles ...

fawick, 2015-06-01:
From a user's perspective, I would choose the following approach: 

1. If Game Detection is disabled, the usual behaviour is kept.
   * Start button starts tracking
   * Stop button stops tracking

2. If Game Detection is enabled, then the semantics of the Start/Stop Buttons change.
   * Start button is re-labelled to "Start Game Detection"
   * When clicked, it Starts Tracking automatically based on the executables as implemented. 
   * Stop button is re-labelled to "Stop Detection"
   * When clicked, tracking (if active) is stopped, and the game-detection is stopped as well.

3. Game-Detection can be enabled/disabled at program start with a command line parameter (`--game-detection` and `--no-game-detection`). In case none of the parameters is given, the last setting of the game-detection checkbox is used (similar to "last profile")

4. The `--autostart` command line option triggers the same action as the Start button, corresponding to the upper three points. In that case, the `<profile>` parameter of `--autostart` is removed and the last active profile is always used.

5. The game detection list can be ordered by the user and detection follows that order. The first executable on the list that is found running wins.

MathijsG, 2015-06-01:
Why would there be needed a better solution if the user manually disables the tracking? People only do that if they want to change anything or want to stop the tracking at all.

sthalik, 2015-06-01:
@fawick that complicates the logic of the main UI a fair bit. Game detection is an optional, minor feature and then it'd need be special-cased for in handlers for starting tracking, etc.

Would it be fine if game detection stopped itself working on manual "Stop" until the last-used game restarted? It stores the current executable internally and can use it for logic.

fawick, 2015-06-01:
Of course there is always the option to run the detection as an external wrapper program. In that case the logic in the main program would be left untouched. Simply run a QProcess with opentrack.exe and use --autostart

sthalik, 2015-06-01:
@fawick I wouldn't recommend that, it can't possibly close open dialogs gracefully. It can start cleanly but can't stop cleanly.

MathijsG, 2015-06-01:
I think it is okay if the feature stops until the last game closed or restarts, because there is a reason for the user to manually stop the tracking in between.

fawick, 2015-06-01:
Given that all destructors are written correctly, Qt applications will handle SIGTERM nicely.

sthalik, 2015-06-02:
It doesn't solve the problem of wanting to edit profile configuration despite game detector and games running. It just kills the app with no SIGTERM handler.

I'm going to release an update for game detection soonish. It rarely resulted in an executable list due to aforementioned bug.

@fawick  There's no implementation for listing Linux processes. It could though be simplified to reuse the decision logic and only specify executable list depending on OS. We could play nicely with Wine here, ps(1) returns Windows executable name instead of wine-preloader, that would help a lot.

sthalik, 2015-06-06:
Are we good now, with exception of lack of process list for Linux builds?

fawick, 2015-06-08:
For what it's worth, I found rc12p1 to be running as expected and without hiccups. Auto-starting it on system bootup and using Game Detection perfectly satisfied my personal use case. In hindsight, I now consider the --autostart command line parameter I suggested as obsolete. Game Detection works for me.

closed 2015-06-15 by sthalik