opentrack/opentrack #162: [Feature request] Bind specific key to profile

MathijsG, 2015-05-24:
While in a game it could be a handy option to bind a key (like you can bind the home and 'end' keys to things as centering and starting and stopping tracking) to switch to a profile. In some games you want to switch to a different profile with different curves in a split second. For example, I want a different sensitivity in ArmA III when I'm on foot and when I'm in a helicopter.

sthalik, 2015-05-26:
We don't have clear semantics for that. Profiles specify trackers and protocols used. To make it work, we'd need to assume these settings to be ignored and overwritten by the previous config used.

We'd also need to notify running trackers, protocols and filters of the profile change to make them reread the settings. There's no software infrastructure for that now.

The binding list would be per-game, so per-profile but then it needs duplicating across all keybinding profiles to make switching work. It's not clear how to make it easy to use.

MathijsG, 2015-05-26:
In my noob opinion it would be something like copying a profile, change things in it and save it under a different name. The keybinding for changing to that profile would be something that stops the current tracking like pressing on the Stop button, then selecting the other profile and then start button again.

sthalik, 2015-06-03:
Now that we have game detection, do you still need this feature?

MathijsG, 2015-06-03:
'need' is a big word, but in many games you want to somehow want to toggle the overall sensitivity with the click of a button, just like some computer mouse have. Or using different axes while on foot instead of in the air. I don't know what would be the best way to achieve such thing?

sthalik, 2015-06-06:
We wouldn't want to stop and start tracking in this case. It will cause a second or two delay before camera is operational. It'll also invoke "center on startup" machinery. The code makes an assumption that "tracking start" is that, not a "switch".

If filter sensitivity is an issue here, we should fix the filter to work for both infantry and airborne cases. There are some filter improvements in unstable that need further confirmation from users with different devices. Right now I don't need either smoothing or deadzone options anymore. The question is whether the new filter gain code works as well for other sensitivity settings.

MathijsG, 2015-06-06:
It's not really a filter issue but more of a overall axis sensitivity thing what I mean. In my case it is mainly this:

When I play ARMA III in a chopper, I can see from the cockpit where my front is, so I don't need a deadzone for pitch and yaw. While on foot, my head view can be on a different angle than my feet are for walking, so I often use a profile which has a dead zone for yaw, so I can "lock" in my front position without keeping my head very still.

I hope this example explains it to you :). In the current situation I just switch from profile when I exit a helicopter, but when you are in the heat of the moment and have no time for this this is no option, and then I just disable the headtracking on foot with the stop tracking shortcut (End)

DerangedPineapple, 2015-06-10:
This change profile on the fly would be very useful feature especially for Arma 2/3 players who fly helicopters/jets and play as an foot soldier in the same game session. 

If we could change at least curves with an button/key press when switching from helicopter/jet to infantry then there would be no need to alt-tab to desktop and manually change profile in use.

sthalik, 2015-07-19:
Confirm whether with Accela improvements this is still an issue.

MathijsG, 2015-07-19:
It's not really an Accela issue, like I said before. It's more of a on-the-fly-sensitivity-change feature which I want.

For example, I'm in a plane in WarThunder. I aim down my sights using the Z axis. Now when I move my head, I really want a different sensitivity than when I was zoomed out, because otherwise the aiming becomes too responsive. So at this point I want a mapping curve which is much less sensitive. Or if you know a better solution for this, but this is the point. Different profiles/sensitivities/curves per game, with easy switching between them.

sthalik, 2015-07-19:
Ideally we'd like to have sensitive tracking without filtering not being sufficient. Or do you wish to change other settings, not filtering?

MathijsG, 2015-07-19:
The filtering isn't the problem in my case. I only use the filtering for filtering out my idle movement of the clip. It's that I need different responsiveness curves of the mapping within the same game. Like a DPI switch on your mouse, but then in OpenTrack somehow. I guess TrackIR has this too, because it's a common issue.

sthalik, 2015-07-19:
We can probably even do joystick buttons. Just need to bind arbitrary keystrokes, not from combobox list. It's been unresolved for too long.

sthalik, 2015-08-22:
I won't be doing this, it's too obscure a use-case.

closed 2015-08-22 by sthalik

MathijsG, 2015-08-22:
Is it too hard to implement? Because in terms of usecases I find it a very logical one. Many people using TrackIR or Opentrack are playing games where they use the joystick the most, and sometimes don't even have the ability to quickly reach to a keyboard. Centering with a joystick button could come in quite handy then :).

sthalik, 2015-08-22:
Centering with a joystick button will be implemented. Switching profiles through a keystoke won't be implemented.