opentrack/opentrack #163: Win 10 build 10074 headtracker crashes with access violation

zazzn, 2015-05-24:
Trying to get this to work on Win 10. using FacetrackNOIR, the crash occurs when you try to start tracking. With Open Track it's the same. If you need access to a dev environment,  Please let me know and I'll be happy to share access to my VM environment. 

Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x0851b648
Faulting module name: ntdll.dll, version: 10.0.10074.0, time stamp: 0x553ac461
Exception code: 0xc0000005
Fault offset: 0x0005cf6e
Faulting process id: 0x1b44
Faulting application start time: 0x01d095fb93676a73
Faulting application path: C:\Users\zazzn\Desktop\Elite Dangerous\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 687ea923-35c6-4a4d-b69c-ef61c89f43d1
Faulting package full name: 
Faulting package-relative application ID:

sthalik, 2015-05-24:
Do you have gdb, the GNU debugger? It's normally built using GNU toolchain. I can send you a version with full debug info.

zazzn, 2015-05-24:
Sure, please drop me a link.... I''m a SE by trade so i'm very savvy technically but have not done any programming since highschool/college. I've played with WinDBG in the past, but a link with some instructions would be very helpful!! Only languages I've written in was watcom pascal and watcom basic, so I have a very limited knowledge outside of your batch scripting/perl scripts ect.

zazzn, 2015-05-24:
Alright, I've verified, for sure it's the FaceTrack API that's crashing by running the Demo from seeingmachines. I noticed on their website, they don't offer the software or API anymore, so I'm betting they don't want to share... (time for them to make money I guess)

I'm not sure what we can do, but I don't believe the source code was ever released for this. That would suck if we had to find a opensource facetracking tool. I'll run the debugs still.

sthalik, 2015-05-25:
Actually the HT face tracker is a portable "replacement" for FaceAPI.

I'll run Win 10 under a VM if you don't have GDB set up.

sthalik, 2015-05-26:
I have a different issue. The PS3 Eye driver spins all CPU when opening the camera in the HT tracker. Attaching a debugger makes it work. There was one random crash like yours.

zazzn, 2015-05-27:
can you send me the version with full debug, I have the debugger installed and i'm with one of my developers at work :)

ufo_10 at hotmail.com

sthalik, 2015-06-01:
@zazzn opentrack-2.3-rc12 build now contains headtracker-ftnoir.exe with DWARF debug info.

zazzn, 2015-06-01:
Downloading now for test ;)

zazzn, 2015-06-07:
I don't think i'm advanced enough to debug sadly :( been trying for the last little while. 

Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x08382038
Faulting module name: ntdll.dll, version: 10.0.10074.0, time stamp: 0x553ac461
Exception code: 0xc0000005
Fault offset: 0x0005cf6e
Faulting process id: 0x16cc
Faulting application start time: 0x01d0a13adffcbbbe
Faulting application path: C:\Users\zazzn\Desktop\Elite Dangerous\opentrack-2.3-rc12\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: a8bf23ec-e861-416d-8082-825682dbfc5a
Faulting package full name: 
Faulting package-relative application ID:

zazzn, 2015-06-15:
Don't know what i'm doing seems too advanced for me.

Did you happen to give it a try at all? Doesn't work in 10130 Win10 either.

sthalik, 2015-06-15:
Try this build and see if it crashes: https://www.dropbox.com/s/p1hntf5vhsvynxt/headtracker-demo-test-20150615.7z

If it crashes, see if there's any debug info in the console window that pops up.

zazzn, 2015-06-15:
This works and it there was tracking!

zazzn, 2015-06-15:
Actually it only crashes, when i try to close it the demo that is.

zazzn, 2015-06-16:
Hi Sthalik,

The demo works, but if I try to use these files in the latest RC replacing the headtrack dir in the latest Opentrack RC nothing shows up and it doesn't seem to launch. 

Suggestions?

sthalik, 2015-06-16:
That demo doesn't include headtracker-ftnoir.exe. Try this one: https://www.dropbox.com/s/ba7or7eql1qoamz/opentrack-test-20150616.7z

zazzn, 2015-06-16:
You know, I think it's the software protection service that is stopping it in windows 10, when i replace the files in Tracker-HT with the new ones that you gave me, I see the following messages below.

I've ran the demo exe alone and it seems to wrok.

The Software Protection service is starting.
Parameters:caller=svchost.exe

Initialization status for service objects.
C:\WINDOWS\system32\sppwinob.dll, msft:spp/windowsfunctionality/agent/7.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:rm/algorithm/inherited/1.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:rm/algorithm/phone/1.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:rm/algorithm/pkey/detect, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:spp/ActionScheduler/1.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:spp/TaskScheduler/1.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:spp/statecollector/pkey, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:spp/volume/services/kms/1.0, 0x00000000, 0x00000000
C:\WINDOWS\system32\sppobjs.dll, msft:spp/volume/services/kms/activationinfo/1.0, 0x00000000, 0x00000000

The Software Protection service has completed licensing status check.
Application Id=55c92734-d682-4d71-983e-d6ec3f16059f
Licensing Status=
1: 05e80e9f-93e3-4433-8b6d-bac4ae66d7bc, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
2: 0cdc4d08-6df6-4eb4-b5b4-a373c3e351e7, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
3: 0eb6e7c2-e5ff-4161-acbe-aec0c135de97, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
4: 29d0b60f-66da-4858-bcaf-9eb513cd310d, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
5: 2d8f6823-fa9f-4d3f-bb16-05c475ca97cd, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
6: 32abc017-5c25-4c88-acf5-137e4e5091bf, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
7: 354d964a-56e7-43c5-a93f-287a7a750bd4, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
8: 4614b66f-a1d7-441c-9731-23f22d0ff4e5, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
9: 4c64cf62-f2d1-4ee6-98a7-bfda0e01c4aa, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
10: 507660dd-3fc4-4df2-81f5-b559467ad56b, 1, 1 [(0 [0x00000000, 1, 0], [(?)( 1 0x00000000)(?)( 2 0x00000000 0 0 msft:rm/algorithm/hwid/4.0 0x00000000 0)( 8 0x00000000 2015/10/01)(?)( 10 0x00000000 msft:rm/algorithm/flags/1.0)(?)])(1 )(2 )(3 )]
11: 51b43152-a885-432f-a1cb-a575ca416784, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
12: 5ce7f628-2651-42fc-9d9b-51a89dc30ae3, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
13: 619e2810-be81-4a4c-b936-07f5ce036c31, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
14: 6d019696-bbb9-42cf-a2dd-1ac564a3cdb9, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
15: 6f45efa9-4271-4ce8-b9e8-7c58fb98d847, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
16: 763fe1c2-d40f-4f3c-9a28-7a3af6f0d987, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
17: 7f71c90a-6279-4274-b398-37f7e9019dd4, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
18: 8da2dfae-e4f5-4e6a-9272-96f8470e033e, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
19: 99cb2a2c-f7ee-48e9-98b0-ae72906c6d45, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
20: 9e473b6d-b591-4c46-9c44-90a865f22e76, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
21: a40c3047-54c3-4bf9-8f09-fb7146cfcb02, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
22: a4383e6b-dada-423d-a43d-f25678429676, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
23: b1154193-7855-4236-838d-f6fc6d161cfb, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
24: b753d452-dc93-4333-90cc-841f0e21c946, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
25: bbc56067-37f8-49dd-87b2-a418a9ba130a, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
26: c06b6981-d7fd-4a35-b7b4-054742b7af67, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
27: d03e843c-9044-4cd4-b5eb-78a9586b5598, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
28: dc3de9b3-9288-486b-a435-3b8f6852bd2d, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
29: ffdd27bd-30ad-4b58-ba24-c7e3b8026443, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
30: 4a8149bb-7d61-49f4-8822-82c7bf88d64b, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
31: b12aee12-c889-471a-b37e-208afdbb86d5, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]
32: ec67814b-30e6-4a50-bf7b-d55daf729d1e, 1, 0 [(0 [0xC004F014, 0, 0], [(?)(?)(?)(?)(?)(?)(?)(?)])(1 )(2 )(3 )]

The Software Protection service has started.
10.0.10130.0

zazzn, 2015-06-16:
That does not work. 

Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x557e6009
Faulting module name: ntdll.dll, version: 10.0.10130.0, time stamp: 0x55602b04
Exception code: 0xc0000005
Fault offset: 0x0005a04e
Faulting process id: 0xb1c
Faulting application start time: 0x01d0a7fc10dbd139
Faulting application path: C:\Users\zazzn\Desktop\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 5f76362f-66e0-466c-94d2-f1c7537eabf8
Faulting package full name: 
Faulting package-relative application ID:

zazzn, 2015-06-16:
Fault bucket 128856585189, type 5
Event Name: FaultTolerantHeap
Response: Not available
Cab Id: 0

Problem signature:
P1: headtracker-ftnoir.exe
P2: 0.0.0.0
P3: 557E6009
P4: ffffbaad
P5: 
P6: 
P7: 
P8: 
P9: 
P10: 

Attached files:
C:\Windows\ServiceProfiles\LocalService\AppData\Local\Temp\FTH316B.tmp\fthempty.txt
C:\Windows\ServiceProfiles\LocalService\AppData\Local\Temp\WER317B.tmp.WERInternalMetadata.xml

These files may be available here:

Analysis symbol: 
Rechecking for solution: 0
Report Id: 62f59c5b-13ef-11e5-b691-bcaec5cef5ed
Report Status: 8
Hashed bucket: 4e9a1a5838bde5360dd16d1574137a59

Fault bucket 107401890294, type 1
Event Name: APPCRASH
Response: Not available
Cab Id: 107375590297

Problem signature:
P1: headtracker-ftnoir.exe
P2: 0.0.0.0
P3: 557e6009
P4: ntdll.dll
P5: 10.0.10130.0
P6: 55602b04
P7: c0000005
P8: 0005a04e
P9: 
P10: 

Attached files:
C:\Users\zazzn\AppData\Local\Temp\WER31A7.tmp.WERInternalMetadata.xml
C:\Users\zazzn\AppData\Local\Temp\WER32E1.tmp.appcompat.txt
C:\ProgramData\Microsoft\Windows\WER\Temp\WER32F1.tmp.hdmp
C:\Users\zazzn\AppData\Local\Temp\WER33BD.tmp.WERDataCollectionFailure.txt

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_headtracker-ftno_4efda87b437575ebd57c906bace4388e4d0e3ae_8af91e4e_cab_1e4053c6

Analysis symbol: 
Rechecking for solution: 0
Report Id: 5f76362f-66e0-466c-94d2-f1c7537eabf8
Report Status: 8
Hashed bucket: ef0728a07e8a49490c1269e57bb88c29

sthalik, 2015-06-16:
Try to reproduce with different camera.

zazzn, 2015-06-16:
@sthalik Ok! seems working on my Lenovo T450, on build 10074 of win10.

Guess the PS3EYE is not working for some reason. Will play with it some more, any ideas why? Do you believe the driver could be at fault? The driver seems to work when I use the CLEYE app. Also the demo you sent worked.

zazzn, 2015-06-16:
@sthalik Verified, PS3 EYE is causing the crash now. Attached to the T450 and tested the bulit in camera, and it works, then tried the PS3EYE with CLEYE drivers, and it crashes when trying to use Headtrack-ftnoir.exe The demo works with the PS3EYE that you sent earlier though.

Do you have this hardware? Do you want to test with this hardware? If so, are you in the US? I'll send you a new one from amazon. it's only 9$ ;) I can always return and get another cam, but this has high FPS for tracking which is why I purchased it.

sthalik, 2015-06-16:
I think the issue is that this is a console application that doesn't process any GUI events. Adding a nonexistent GUI just to appease CL-Eye seems in order. I have this webcam, just need to wait till i get a win10 VM again.

zazzn, 2015-06-16:
@sthalik Do you have team viewer? I'll send you a session code you can just use my VM environment, if you would like. I have it snapshotted, for every build recently.

sthalik, 2015-06-19:
@zazzn I have my own environment. Since there were problems on my Windows 10 VM too, I'll get to you once there's something that works for me at least.

zazzn, 2015-06-21:
@sthalik  any idea when you may get this fixed. I'm debating returning the cam. I got it from amazon, and I'll just return it and get a diff one, but I know since this had a high FPS it would be a good cam to use.

sthalik, 2015-06-21:
@zazzn try this one https://www.dropbox.com/s/4lniiarv6ve2f6d/tracker-ht-20150621.7z

Works fine on my end.

But PS3 Eye has awful image quality and it's only good for LED tracking.

zazzn, 2015-06-23:
@sthalik Doesn't seem to work, I used the whole installer, that you provided before, replaced the tracker-ht dir with the one in the above download and nothing happends when I run the head tracker. 

I know it has horrid quality but seemed to track my face perfectly well in all the demos.

zazzn, 2015-06-23:
Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x73657922
Faulting module name: ntdll.dll, version: 10.0.10130.0, time stamp: 0x55602b04
Exception code: 0xc0000005
Fault offset: 0x0005a04e
Faulting process id: 0x26cc
Faulting application start time: 0x01d0ad6132615322
Faulting application path: C:\Users\zazzn\Desktop\Elite Dangerous\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 765a97f5-0807-4cde-97ee-578317e4ce0b
Faulting package full name: 
Faulting package-relative application ID:

zazzn, 2015-06-23:
here's from the event logs.

Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x73657922
Faulting module name: ntdll.dll, version: 10.0.10130.0, time stamp: 0x55602b04
Exception code: 0xc0000005
Fault offset: 0x0005a04e
Faulting process id: 0xfb0
Faulting application start time: 0x01d0ad622b207ca9
Faulting application path: C:\Users\zazzn\Desktop\Elite Dangerous\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 1ce325bc-8139-41dd-9718-c344ee0c784a
Faulting package full name: 
Faulting package-relative application ID: 

Fault bucket 107416573519, type 1
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: headtracker-ftnoir.exe
P2: 0.0.0.0
P3: 73657922
P4: ntdll.dll
P5: 10.0.10130.0
P6: 55602b04
P7: c0000005
P8: 0005a04e
P9: 
P10: 

Attached files:
C:\Users\zazzn\AppData\Local\Temp\WER5654.tmp.WERInternalMetadata.xml
C:\Users\zazzn\AppData\Local\Temp\WER579D.tmp.appcompat.txt
C:\ProgramData\Microsoft\Windows\WER\Temp\WER57AE.tmp.hdmp
C:\Users\zazzn\AppData\Local\Temp\WER584B.tmp.WERDataCollectionFailure.txt

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_headtracker-ftno_9625607085bb865183c240ffd3731eb8ddd3ab82_db306743_01835867

Analysis symbol: 
Rechecking for solution: 0
Report Id: 1ce325bc-8139-41dd-9718-c344ee0c784a
Report Status: 4104
Hashed bucket: cdc1afe01c5cd76b6f99e6d314042dc1

Faulting application name: headtracker-ftnoir.exe, version: 0.0.0.0, time stamp: 0x73657922
Faulting module name: ntdll.dll, version: 10.0.10130.0, time stamp: 0x55602b04
Exception code: 0xc0000005
Fault offset: 0x0005a04e
Faulting process id: 0x2a00
Faulting application start time: 0x01d0ad624147aa8b
Faulting application path: C:\Users\zazzn\Desktop\Elite Dangerous\install\tracker-ht\headtracker-ftnoir.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: 2f71de66-f979-48a0-bf52-70979adaf24c
Faulting package full name: 
Faulting package-relative application ID: 

Fault bucket 107416573519, type 1
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: headtracker-ftnoir.exe
P2: 0.0.0.0
P3: 73657922
P4: ntdll.dll
P5: 10.0.10130.0
P6: 55602b04
P7: c0000005
P8: 0005a04e
P9: 
P10: 

Attached files:
C:\Users\zazzn\AppData\Local\Temp\WER7805.tmp.WERInternalMetadata.xml
C:\Users\zazzn\AppData\Local\Temp\WER78E1.tmp.appcompat.txt
C:\ProgramData\Microsoft\Windows\WER\Temp\WER78E2.tmp.hdmp
C:\Users\zazzn\AppData\Local\Temp\WER797F.tmp.WERDataCollectionFailure.txt

These files may be available here:
C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_headtracker-ftno_9625607085bb865183c240ffd3731eb8ddd3ab82_db306743_1913798c

Analysis symbol: 
Rechecking for solution: 0
Report Id: 2f71de66-f979-48a0-bf52-70979adaf24c
Report Status: 4104
Hashed bucket: cdc1afe01c5cd76b6f99e6d314042dc1

zazzn, 2015-06-26:
@sthalik 

Can you explain how you got yours working? I can't seem to get mine working what is going wrong? I'm on build 1130 now.

zazzn, 2015-06-27:
@sthalik any updates? The version you posted in dropbox doesn't work. It still crashes, I'm using headtrack FT noir not facetrack!

sthalik, 2015-06-28:
@zazzn I can't help you without debug information. Fire up "gdb", it's not hard to use.

sthalik, 2015-07-16:
Try to confirm rc14 if whole opentrack process crashes.

sthalik, 2015-07-22:
I'm closing this due to lack of response. We'll reopen if you get back on it.

closed 2015-07-22 by sthalik

zazzn, 2015-07-22:
Downloading now.

zazzn, 2015-07-22:
Looks like it's working!

sthalik, 2015-07-22:
I'm glad to hear that!

There need to be more changes for HT beforee 2.3 release. It performs badly from my experience, especially on high yaw.

zazzn, 2015-07-22:
Ah, i changed the resolution to 640x480 and seems to track much better.

I'm buying my time with this until the occulus comes out in xmas and I can use it with elite dangerous ;) 

Don't want to buy a 3rd monitor nor do I want to use IR unless i buy the IR from ebay... issue is the ps3 eye i have is new so I'm betting it has the IR filter on it.

sthalik, 2015-07-22:
You can try the Aruco paper marker tracker, it works much better than face tracking. I'm using it exclusively with a 30 FPS webcam and having a blast. All it needs is a b&w printer. See the wiki page.

zazzn, 2015-07-22:
hrm, i will do that for sure!! thanks,

zazzn, 2015-08-22:
Ok, just downloaded the latest build and the head track seems to go nuts. Can't track the face but worked really really well with the demo? 

Aside from the print out that makes things easy, (paper tracker) can face tracker not track as well as the demo?

sthalik, 2015-08-22:
There's something bad going on with it. I agree. We'll either replace the algorithm entirely or make something bearable at least.

See #197