opentrack/opentrack #164: Pitch changes too much when moving head left to right. Point tracker.

frost555, 2015-05-26:
I have an issue with 2.3 version. When i move my head left pitch goes negative, when i move my head right pitch moves positive. The difference is about 30 degrees, although i try to keep my head level. I dont have the same issue with 2.2 version, though my settings are almost identical. I made a video to demonstrate the effect. Maybe my settings are wrong? Or is there a problem with software?

http://www.youtube.com/watch?v=ek8489xi-_o&feature=youtu.be

frost555, 2015-05-26:
Issue fixed by unticking center on startup in keyboard settings

sthalik, 2015-05-26:
You neutral camera pose is rolled by 20 degrees from the video. Maybe we should add an option for camera roll angle?

frost555, 2015-05-27:
If it is possible to accurately solve the geometrical problem of tracking with various model orientation at neural pose,  implementing those techniques would be great.

MathijsG, 2015-05-27:
Or make an option for clip angle. Many headphones don't have straight vertical lines to attach the clip to. My clip defaults to this, while my camera is perfectly straight:

[image]

Somewhere OpenTrack still assumes the position should be 3 crosses perfectly in a vertical line.

When I move my head upwards, it incorrectly adds yaw to the pitch movement, which is a pretty annoying flaw, which FaceTrackNoIR doesn't have.

[image]

[image]

My headphone is a SHP2000 with no straight vertical edges :(.

sthalik, 2015-05-29:
@MathijsG use custom model coordinates for slanted angles. We already have plenty of rotation options, and a new one for that is redundant.

frost555, 2015-05-29:
I think that is a little bit too hard for average user. As i understand i need to use rotation matrixes to calculate custom model dimensions. Maybe we need process similar to existing calibration. Press the button and you will have custom model dimensions for the angles at default pose.

MathijsG, 2015-05-29:
FaceTrackNoIR has no problems with rotated angles when the model dimensions are correctly entered (which in case are, in both programs) why does have OpenTrack a problem with it? Manually entering those coordinates would be a pain in the ass, because you can't decently measure it, the model dimensions itself are just regular clip. I thought all calculations were based on the position of the clip when you have centered it, with the button. But it looks like many calculations are still being made based on a perfect vertical position of a clip.

sthalik, 2015-05-31:
@frost555 you don't need matrices, they're too verbose. I won't provide a trigonometry solution since it can't be expected of users to perform.

Current rotation code has an interesting property where rolling the model by 90 degrees swaps pitch and yaw. I understand this property is less important than your common use-case. I'd like to release a fix after confirming it doesn't break FreePIE-equipped inertial trackers.

sthalik, 2015-06-01:
The feature's been committed for the next RC. Please provide feedback.

MathijsG, 2015-06-01:
With the current RC (12) the unwanted yaw is fixed again and my clip stays horizontal again while looking up :-), thanks for fixing. The auto center on starting of a profile is also a very welcome feature.

Your software is quickly getting better and better :)

The profile binding to executables is also very nice. Is it suppose to also start the headtracking itself? Or just selecting the profile? I didn't get any auto starting, will try further with the new version.

sthalik, 2015-06-01:
If you enable the checkbox and executable name is the same, it should switch to that profile and start tracking.

MathijsG, 2015-06-01:
Ah, didn't recognize the checkbox. Now it is working fine. Very nice!

sthalik, 2015-06-01:
@frost555 can you confirm it works now?

@KyokushinPL responded that the change doesn't break phone tracking.

frost555, 2015-06-01:
Yes, i can confirm that i no longer have the reported issue. Thanks a lot!

closed 2015-06-02 by sthalik