opentrack/opentrack #165: 2.3rc10 Not working in FreeTrack compatible game (GP Bikes)

HornetMaX, 2015-05-27:
Hi, I'm new to OpenTrack, great project !
I landed here when reading about EDTracker (http://www.edtracker.org.uk/) as I may be interested for Piboso's simulators, GP Bikes, MX Bikes, Kart Racing Pro and World Racing Series (all are FreeTrack ready and allow mouse-look).

I have no tracking device yet, so I tried using head tracking (via webcam) as input. Everything seems to work fine in the OpenTrack GUI, but I get nothing in-game if I use FreeTrack output. Mouse emulation works a bit better but only in game menus, no in-game (as described in issue #157).

I checked that the game does work with FreeTrack (FreeTrack v2.2, demo mode): pass.

I have downloaded and installed 2.3rc10, run it, cofigure and start the tracking, then run my game. Am I missing something ?

MaX.

P.S.
If anybody wants to give it a try all the 4 games are downloadable and have a free demo mode that allows you to play (only game mods and extra bikes/cars/tracks are not allowed). You can find them here:

http://www.gp-bikes.com/
http://www.mx-bikes.com/
http://www.kartracing-pro.com/
http://www.worldracingseries.net/

MathijsG, 2015-05-27:
Did you try different settings on the Freetrack protocol? Click the button with the 3 dots ( ... ) on the protocol section for some options, in some games you need to act like TrackIR and in some there is official support for freetrack.

rizer1980, 2015-05-28:
Not working with IL2 Battle of Stalingrad.
Try working config, default, aloso try different option in "freetrack protocol settings"
2.3.9 work fine.

HornetMaX, 2015-05-28:
I tried to play around with the settings but no joy. Will try more tonight.

I have tried FaceTrackNoIR (v1.70) and FreeTrack (v2.2, demo mode): they both work fine with the above games, without any particular messing with the settings :(

MaX.

sthalik, 2015-05-29:
@rizer1980 there were no changes from rc9 with regard to freetrack protocol.

Do you have opentrack installed in user directory or in Program Files? Filesystem permissions can respond differently to these directories.

HornetMaX, 2015-05-29:
@sthalik thanks for the advice. I simply put opentrack on the desktop, will try to move it around (Program Files, c:/opentrack, etc) and report back. The only unusual thing is that my "Desktop" is not in  the usual location (e.g. c:\Users\MaX\Desktop) but moved to another drive.

rizer1980, 2015-05-29:
2.3.10 in directory "program files", same as 2.3.9.
Try also "desktop" folder, nothing changed.
Check permisiion for files, all allowed.
Try run as "run from system administrator".
Try to copy one file, opentrack.exe ver 2.3.10 to folder with software opentrack 2.3.9, 
and run it, not working.

no idea more... (

HornetMaX, 2015-05-29:
Tried with install directory just under c: or in c:\Program Files (x86), no joy: still nothing happens in-game.

Is there a 2.3-rc9 build I can try ? (I can only see the sources here for rc9).

rizer1980, 2015-05-29:
@HornetMaX 
2.3.9
https://drive.google.com/file/d/0B3gQUxLbCRBgV0JkcWNJMGw2Qkk/view?usp=sharing

HornetMaX, 2015-05-30:
2.3rc9 behaves just like 2.3rc10 for me :(

One question: the file clientfiles/freetracktest/freetracktest.exe is supposed to do something when opentrack is runing (with freetrack output) ? because mine does nothing (rc9 and rc10, DLL loaded = false, no numbers moving).

On the other hand, if I run FreeTrack and then freetracktest.exe, then it shouws dll loladed = true and the numbers move.

MaX.

sthalik, 2015-05-30:
@HornetMaX you may have "use only TrackIR" selected in freetrack 2.0 protocol settings window. It works for me in newest build.

rizer1980, 2015-05-31:
ok, it is working for now. )
I tried freetracktest.exe, no dll loaded = false at first, run 2.3.9 and freetracknoir several times.
And after that dll = true, but no numbers under 2.3.10.
After that i place 2.3.10 to c:\opentrack directory and reboot just in case.
And 2.3.10 work for now, only in \opentrack directory, not working in program files, and not working in desktop.
freetracktest also working now.

thx.

HornetMaX, 2015-05-31:
It's driving me nuts :)

@sthalik : No, I haven't selected "use only TrackIR" (I use "Enable both" or "use FreeTrack, hide TrackIR").

What I do:

1. Cleanup/uninstall everything about FaceTraclNoIR, FreeTrack and OpenTrack, then reboot
2. Unzip the content of opentrack-2.3-rc10.zip into c:\opentrack
3. Run opentrack.exe, configure it (ht input, freetrack output, configure keys, save to .ini file) and start tracking
4. Run freetracktest.exe

No joy, not even if running both exes as admin :( 

DLL loaded = false 

I'm on Win 7 x64 (ultimate), if that matters.

MaX.

sthalik, 2015-06-01:
@HornetMaX okay. Start regedit... HKEY_CURRENT_USER\Software\Freetrack\FreetrackClient

The "Path" key on the right should have a directory where freetrackclient.dll is stored.

If it exists on your system, it's a harder matter to analyse though.

HornetMaX, 2015-06-01:
AAARGGH  !!!!

Office laptop (win7 x64): clean up everything, install rc12 (on Desktop), run opentrack.exe, check registry key, run freetracktest.exe --> OK, WORKS !!

Hoe PC (win7 x64): clean up everything, install rc12 (on Desktop), run opentrack.exe, check registry key, run freetracktest.exe --> DLL loaded = false :( :(

MaX.

sthalik, 2015-06-01:
Did you verify the .dll exists as per my last comment?

HornetMaX, 2015-06-01:
@sthalik : yep, everything seems to be OK. The registry key gets correctly set. 

I tried to clean up everything (even the registry under HKEY_CURRENT_USER\Software\OpenTrack-2.3), then reinstall. HKEY_CURRENT_USER\Software\Freetrack\FreetrackClient points to the right directory, but freetracktest.exe doesn't find the .dll.

I even switched forward/backward slashes in the registry path, out of desperation :)
I've even checked I have no other freetrackclient.dll hanging around (except in the installation directory).

It's just puzzling. On my office laptop it works fine now (well, the freetracktest.exe works, in-game I still see nothing happening, but that coul dbe the game's fault)..

[but at least mouse emulation now works]

sthalik, 2015-06-02:
@HornetMaX there's a tool called sxstrace.exe that can help. It's sadly hard to use. Also there's procmon.exe from sysinternals that can trace freetracktest.exe. Also a simple tool can be written to try open freetracktest and report any errors along the way.

It's tedious but I can't reproduct the issue on my end, so can't really share any particular expertise. I'll try to provide some diagnostic tool in time but please try for now other methods.

HornetMaX, 2015-06-02:
@sthalik OK, thanks !
I've run Process Monitor on my office laptop (where freetracktest.exe works) and exported the events (filtering on Process = freetracktest.exe). I'll do the same at home and see if I get something strange.

HornetMaX, 2015-06-02:
@sthalik I run procmon on my office laptop (freetracktest.exe working) and on my home pc (not working). I saved the output (filtering only stuff relative to freetracktest.exe), I've put the logs here: https://drive.google.com/file/d/0BzmU7Qoo77i1X0h5azBBRXBxRzA/view?usp=sharing

Honestly, they are not very useful to me :)

HornetMaX, 2015-06-02:
Wow, I thikn I've found it. If MSi Afterburner (with its RivaTuner Statistics Server) is running, freetracktest.exe does not work. Weird !

MaX.

sthalik, 2015-06-02:
@HornetMaX for me it worked with MSI Afterburner always. Strange indeed.

HornetMaX, 2015-06-02:
@sthalik yeah strange. 

Another random question: is the freetrack 2.0 protocol backward compatible with the original 1.0 protocol ? The games I'm interested into works wit FreeTrack 2.2 demo mode (which I suspect is using the v1.0 protocol). The info around all this is very confusing ...

Enough progress for tonight :)

sthalik, 2015-06-02:
Our protocol can only be used with the data structures for 2.0. In that sense, it's not backward-compatible.

HornetMaX, 2015-06-03:
@sthalik Thanks. I'll see with the dev of the games then.

Let me know if you want this issue closed or kept open to track the conflict with AfterBurner.

sthalik, 2015-06-03:
You can use any implementation of FreeTrackClient for supporting games. There seems to be no conflict.

HornetMaX, 2015-06-03:
OK, now you lost me :)

I have a game that definitely work fine with FreeTrack v2.2 (in demo mode).
Should that game work fine with OpenTrack just because it works fine with FreeTrack v2.2 ?

Also, will the game need to locate and load the FreeTrackClient dll ?
So maybe it's just a matter of putting freetrackclient.dll on the path (or in the game's directory) ...

sthalik, 2015-06-03:
Path to freetrackclient in registry gets adjusted automatically on tracking start as long as you don't disable freetrack in "freetrack 2.0" protocol settings. From that moment, games starting with use that version.

But that .dll isn't compatible if other implementations try use it.

sthalik, 2015-06-06:
> Should that game work fine with OpenTrack just because it works fine with FreeTrack v2.2 ?

Yes, that's the intention behind the protocol.

HornetMaX, 2015-06-06:
OK, some progress.

I have both FreeTrack 2.2 and OpenTrack 2.3rc12 installed (in their own folders under c:\Program Files (x86)).

When the registry Freetrack/FreetrackClient/Path points to where OpenTrack is installed, freetacktest.exe doesn't work (DLL loaded = false) and my game doesn't see any freetrack.
If I stop AfterBurner, freetacktest.exe works but my game does not see any freetrack.

When the registry entry points to where FreeTrack is installed, then everything works fine, including my game. And no matter if AfterBurner is running or not. To do that I have to: 1, start freetrack in demo mode (sets the registry), 2 start my game, 3 stop freetrack and start opentrack.

Weird DLL stuff going on ?

sthalik, 2015-06-07:
> Weird DLL stuff going on ?

I agree. Please try this build:

https://www.dropbox.com/s/cnnmt4mz74ic5ve/freetrackclient.dll

sthalik, 2015-06-07:
I was able to verify it working with arma 2. arma 3 seems to have removed freetrack support...

sthalik, 2015-06-07:
Interesting http://www.free-track.net/forum/index.php?showtopic=4426

> You should activate freetrack in your ARMA 3 profile txt file. It is situated i your doc. directory 
> 1)C:\Users\username\Documents\Arma 3\your ingame name.txt (for default profile) and 
> C:\Users\username\Documents\Arma 3Arma 3 - Other Profiles\your ingame name.txt(for other profiles)
> There you should find a line: freetrack=0 ,change it to freetrack=1, save it

MathijsG, 2015-06-07:
ARMA 3 works perfectly here without modyfiing anything, just enable TrackIR as a controller in-game.

HornetMaX, 2015-06-07:
>>Weird DLL stuff going on ?

>I agree. Please try this build:
>
>https://www.dropbox.com/s/cnnmt4mz74ic5ve/freetrackclient.dll

@sthalik: It works fine with that dll  !! **w00tt!!11**

Game sees OpenTrack's FreeTrack output, freetracktest.exe works too and no conflict with AfterBurner.

One minor quirk: in freetracktest.exe, only the virtual pose numbers do vary, the raw one stay locked to zero (but I had this even with Freetrack's original client dll).

Great !

Let me know what's the plan to include the modified dll in your next releases, I'll inform the people interested on the game forum.

Thanks a lot for the great support !

sthalik, 2015-06-07:
@HornetMaX it's going to be in the next release candidate.

closed 2015-06-07 by sthalik