opentrack/opentrack #166: Freetrack Protocol 2.0

MarceloTomaz, 2015-06-01:
Hello people!
I've been looking Freetrack Protocol 2.0, but found nothing concrete.
I am using an Arduino with a MPU6050 module for headtrack and would like to implement Freetrack Protocol 2.0.

I understand that it takes a server providing data on Freetrack Protocol 2.0 for the games to access the data
Is correct, can anyone help?

Grateful.

sthalik, 2015-06-01:
Do you wish to implement game-side code or tracker-side code?

MarceloTomaz, 2015-06-01:
Tracker-side.

sthalik, 2015-06-01:
Data structures are defined in https://github.com/opentrack/opentrack/blob/unstable/freetrackclient/fttypes.h

Source for freetrackclient.dll is in the same directory. Source for NPClient.dll and NPClient64.dll is available on request.

The .dll runs inside games, to connect to it:

https://github.com/opentrack/opentrack/blob/unstable/ftnoir_protocol_ft/ftnoir_protocol_ft.cpp

The logic includes parsing the .csv file in bin/settings for some game-specific bits. Path to the three .dll's in registry is set there also.

If this seems like too much work, you can also use **libopentrack-proto-freetrack.dll** standalone without involving opentrack. It has some simple methods defined as IProtocol in https://github.com/opentrack/opentrack/blob/unstable/opentrack/plugin-qt-api.hpp

I'm here if you have any additional questions.

sthalik, 2015-06-01:
There was already talk on using opentrack plugins without running opentrack in #151. Maybe this helps too.

MarceloTomaz, 2015-06-01:
Thanks I will study!
My application runs C #, so I want to understand how the protocol.
Sorry, I could include libopentrack-proto-freetrack.dll and call their functions using C # or could still run without GUI OpenTrack?

sthalik, 2015-06-01:
C# can't interface with C++, you need to write C API wrappers to access the library.

Alternatively you can send data through UDP protocol to opentrack.

sthalik, 2015-06-01:
Alternatively you can duplicate https://github.com/opentrack/opentrack/blob/unstable/ftnoir_protocol_ft/ftnoir_protocol_ft.cpp in C# and interface with the memory mapping yourself.

MarceloTomaz, 2015-06-02:
I will use the udp sender to communicate, but I'm sending XYZ, yaw, pitch roll without success. I'm getting Opentrack data indicating communication.
Example: byte [] sendBytes = {0,0,0 , 10,110,50}
Any idea?

Grateful!

MarceloTomaz, 2015-06-02:
Sorry, there was a lack of attention.
I found the error.

closed 2015-06-03 by sthalik

JimmyRaoUF, 2019-08-12:
> Data structures are defined in https://github.com/opentrack/opentrack/blob/unstable/freetrackclient/fttypes.h
>
> Source for freetrackclient.dll is in the same directory. Source for NPClient.dll and [...]

Hi @sthalik , if I want to implement game side receiver, is NPClient.dll related? What's its relation with TrackIR.dll or FreeTrack.dll?

Thank you:)

sthalik, 2019-09-22:
@JimmyRaoUF NPClient is the library for games that support only TrackIR. opentrack creates a key (see https://github.com/opentrack/opentrack/blob/unstable/proto-ft/ftnoir_protocol_ft.cpp#L129) for the library to work at all.

This library version is whitelisted in several anti-cheat programs so I recommend using this binary version rather than writing NPClient.dll from scratch.