opentrack/opentrack #167: Point Tracker no output in stable or 2.3

jadeybaby166, 2015-06-02:
I am attaching a screenshot. Have tried multiple instances run from desktop program files and x86 directories. I presume that there is not supposed to be and installer with it and that it is run straight from the exe in the directory. it runs fine on my laptop running windows 8 and i have previously run it on this computer (windows 7, both 64). I''m totally stumped as to why this isnt working.

[image]

jadeybaby166, 2015-06-02:
[image]

sthalik, 2015-06-02:
Did you input model coordinates in PT settings?

jadeybaby166, 2015-06-03:
if you mean tracker clip dimentions im using the default ones because im using the delanclip. i tried to calibrate it but all went to 999.9. If that isnt waht you mean can you please explain.

sthalik, 2015-06-03:
What are the dimensions you specified then?

Len62, 2015-06-03:
In the settings area of point tracker. There are no default numbers.
[image]

sthalik, 2015-06-03:
Please post your full configuration file.

jadeybaby166, 2015-06-03:
https://www.sendspace.com/file/1scfy3

That is the ini file (i presume thats what you meant).

Also i just downloads a fresh 2.2 install and there are definitely numbers in the model dimentions. see below. the numbers in the ini should match the ones in the screenshot.
i belive the numbers shown in the screenshot above, and mine below defualt to the dimesions of the track clip pro do they not? the clip i have has the same dimentions and is a cheaper sturdier alternative.

Len62, 2015-06-03:
Make sure when you fire-up a new install/version that you load the default.ini file from the current versions drive location. Opentrack, due to info in the registry, will start with an ini file from an older version. Whenever I start fresh I load the default.ini and the model dimensions are always zero.

jadeybaby166, 2015-06-03:
I dont know what happened before, but i just loaded the ini i was using on my laptop, on my desktop and its working now. Im not sure what i was loading before to get the results i was getting but im guessing it wasnt the ini that i was using on my laptop. Sorry to have wasted your time. I'll make sure i have a blank ini whenever i install a fresh one.

sthalik, 2015-06-04:
@Len62 we have a separate registry key for opentrack-2.3.

Glad the issue got resolved.

closed 2015-06-04 by sthalik