opentrack/opentrack #168: Tracking randomly stopping when using auto tracking based on executable

MathijsG, 2015-06-02:
I tried the new very nice feature of automatically starting tracking based on starting a specific executable.

In my case this was Il-2 Cliffs of Dover. When I started the game the auto tracking started perfectly, but after a while it randomly stops tracking for a few seconds and then starts again. When I disabled the auto tracking based on executable feature the tracking didn't stop, therefore I think the error must be somewhere in the executable tracking.

I will check this out more with other games within a few days.

sthalik, 2015-06-03:
There was a bug making it run erratically. It's fixed in rc12p1 binary release.

closed 2015-06-03 by sthalik

MathijsG, 2015-06-04:
Confirmed that this is perfectly working again indeed :).