opentrack/opentrack #169: Game is crashing when selecting EWMA filter, also no octopus with this filter

MathijsG, 2015-06-04:
I'm trying out some filtering and stuff the get a smoother experience, but when I try the EWMA filter the game (IL-2 Cliffs of Dover) just crashes, and I don't see a moving octopus. Is this just me or is this filter deprecated?

sthalik, 2015-06-06:
I can't reproduce the issue. What "game output" values are you getting?

MathijsG, 2015-06-06:
The game output stays zero, while the raw data does give numbers. But as soon as I click start the octopus also hides (that's supposed to work that way?)

And with Kallman I do get the Octopus, but when I want to calibrate any settings of Kalman I just get an empty popup window with no settings.

sthalik, 2015-06-06:
My guess is that there's division by zero resulting in NaN values passed around. Try this build: <<https://www.dropbox.com/s/vhxgn4er967g3yr/libopentrack-filter-ewma.dll>>. If the octopus no longer disappears, it's a proper fix.

sthalik, 2015-06-06:
The binary is larger because there's debug info in it. It's otherwise identical with NaN check applied.

MathijsG, 2015-06-06:
When I put the DLL and run the software the filter isn't even selectable anymore:
[image]

sthalik, 2015-06-06:
My bad, try this whole installation directory: <<https://www.dropbox.com/s/2rm8udnvd2l6qpw/opentrack-ewma-test-20150606.7z>>

MathijsG, 2015-06-06:
Nice it's working again now :-). Kalman only still got no settings, but I think that is working as intended?

All those filters are a little confusing to me though, maybe we can provide some in-program info about which filter is best for which situation or something? I would help with documenting things.

sthalik, 2015-06-06:
Kalman and EWMA are @dbaarda's turf. He works on them when he feels like it. It's mostly for academical purposes that they're in-tree.

I'm working on improving Accela before release and gathering feedback. Let's wait with filter performance evaluation until all that could be done, has been done.

sthalik, 2015-06-06:
Thanks for bug report.

closed 2015-06-06 by sthalik