opentrack/opentrack #170: roll affects pitch after centering

cyboff, 2015-06-05:
Hello,
when I was trying to set up Opentrack for Windows Phone (as UDP sender) together with Opentrack 2.3-rc12p1, I found strange behaviour of Center function:
When I was comparing raw tracker data with game data with no filter, one-to-one linear mapping, and centering disabled after start values from raw tracker and game data are always the same and changes to yaw, pitch and roll are not affecting each other (I have my phone on the table to simply simulate each axis without affecting other). But after centering via keyboard, change in roll axis in raw tracker (and no change in yaw and pitch) also affects pitch in game data and opposite - change in pitch axis (and no change in yaw and roll) in raw data affects roll in game data. Yaw axis works fine. See attached screenshots:

data without center applied:
[image]
[image]

after center and change in roll axis:
[image]
[image]

after center and change in pitch axis:
[image]
[image]

sthalik, 2015-06-05:
There are some unresolved differences with how centering should work. There are two possible solutions:

- after rolling, "up" is opposite of gravity's vector
- after rolling, "up" is where the phone's top points out

We're using the latter behavior.

The FreePIE tracker for Android phones crudely solves that by adding a user-specified value (-90, +90, -180, +180) to each of yaw, pitch and roll. That allows for treating a rotated phone as if it wasn't rotated at preferred position.

We can't enforce the former centering behavior globally as PointTracker users have the same problem as you have with the latter behavior. But adding the same bunch of FreePIE options to the UDP tracker can solve the problem you're having.

cyboff, 2015-06-05:
Thanks for clarification!

sthalik, 2015-06-06:
What position are you centering from?

cyboff, 2015-06-06:
Usually I just need to center yaw (and pitch a little), as I have my phone attached to headphones:
raw data are usually around these values: yaw 110, pitch -10, roll 0
Result is that after centering, change in pitch of my head affects roll game values more than pitch, and
rolling my head is affecting game pitch values more...

sthalik, 2015-06-06:
That follows from my reasoning about centering. Try this version. It's larger than releases because of debug information.

<<https://www.dropbox.com/s/2rm8udnvd2l6qpw/opentrack-ewma-test-20150606.7z>>

cyboff, 2015-06-06:
I can't see any options to change in UDP tracker ( I'm using this: http://www.windowsphone.com/en-us/store/app/opentrack-head-tracking/1c604f32-6d68-40ef-aa44-3163e30f547f )  

also on UDP receiver in Opentracker there is only possibility to specify port number...

Without that, there is no change in behaviour..
But I think version rc10 was working quite well for me... but can't find binary version to download and check again:(

sthalik, 2015-06-06:
Sorry, I updated the wrong file. You should now in the same link, see yaw/pitch/roll options in the UDP tracker.

We can't go back to rc10 behavior since it works badly for PT tracker users.

cyboff, 2015-06-06:
It's much better now :D Thank you very much!

closed 2015-06-06 by sthalik