opentrack/opentrack #171: [Feature request] Bring back possibility to configure an Auxilliary Tracker

jochenschmich, 2015-06-10:
Hi there,

in V2.2 it was possible to configure a secondary tracker. This way you could combine a Gyroscope Tracker (via Joystick for example) and a IR Clip.

In V2.3 this options seems to be gone.
Is there any way to reactivate it?

Cheers

sthalik, 2015-06-14:
The exact functionality of "secondary tracker" won't come back. It was poorly designed.

A better idea is to make a new "Meta" or "Multi" tracker that allows for selecting arbitrary other trackers and assigning axes.

Any objections?

jochenschmich, 2015-06-26:
OK, sounds quite good. Configuration of each individual tracker should not change, though - I mean interface-wise.

Cheers.

jochenschmich, 2015-07-29:
Will this feature be part of v2.3?

sthalik, 2015-07-30:
Yes.

sthalik, 2015-09-07:
How important is running the same tracker twice? It can be done in options.hpp bundle support.

jochenschmich, 2015-10-06:
Might be useful in a scenario where multiple virtual Joysticks are used for example... it's not especially important for me - might be important for other people, though.

cutzmf, 2016-06-29:
Ummm... I'm waiting for this feature.
I appreciate your work =) Thank You.

Ec0nomist, 2017-05-28:
When? :)
https://forum.il2sturmovik.ru/uploads/monthly_05_2017/post-38890-0-66333900-1495974689.png

sthalik, 2017-05-30:
I think this is enough for now. Are there any other use-cases? I'm closing but keep me notified if you need something else.

sthalik, 2017-05-30:
@Ec0nomist I need a second user confirming Vive works in the test build.

closed 2017-05-30 by sthalik