opentrack/opentrack #174: Flight Simulator X Steam Edition stuttery headtracking

MathijsG, 2015-06-17:
When I play Flight Simulator Steam Edition and start the headtracking with the Flight Simulator protocol (Microsoft FSX Simconnect) it gets all slow and jerky in-game. When I alt tab out of the game, and the game goes windowed and smaller it's all smooth, but back in the full game it's jerky again. FaceTrackNOIR has zero problems with this protocol in this game.

If you need more specifics, just ask :).

sthalik, 2015-06-18:
Try pin FSX to all your cores except 0 and 1, then compare with pinned to all cores.

MathijsG, 2015-06-18:
How do I do that? Because the game itself is running rocksolid, and it seems weird to me that FaceTrackNoIR has 0 problems with it at all.

sthalik, 2015-06-18:
Use Windows Task Manager.

MathijsG, 2015-06-18:
It's by default pinned to this:

[image]

Whatever checkbox I select it makes no difference.

It even stutters when the simulation is paused and not on full screen, only when there is another window which has the focus, the headtracking is smooth, otherwise it's stuttery. FaceTrackNoIR: zero problems.

Seems weird, maybe I will record some video when I have the time.

sthalik, 2015-06-19:
It might be because we communicate with SimConnect 250 times a second. Doing it on another thread for a limited amount of time can help. I'll try to get it fixed soon.

MathijsG, 2015-06-19:
Okay I'm willing to test everything out to make this piece of software even better :).

sthalik, 2015-07-03:
@MathijsG  try this one: https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll I have no FSX installed so it could crash on startup though.

sthalik, 2015-07-03:
Try RC13.

MathijsG, 2015-07-03:
I will try asap and report back on all my reported bugs :-). Thanks

MathijsG, 2015-07-03:
The protocol implies a default location for flight simulator. If you then click start you get:
[image]

Maybe a prompt with Where do you have Microsoft Flight Simulator Installed? Or something would be more userfriendly?

And if you click start after clicking OK to the error above you get a full program crash.

Will try now with manually changing the path to the dll in the settings of the protocol.

sthalik, 2015-07-03:
I've ninja-edited the update. It shouldn't crash anymore. In protocol settings you have an option for FSX version. It's more involved than directory name, it uses SxS which deals with application manifests.

MathijsG, 2015-07-03:
Still same problem, stuttery movement while game is perfectly fluid. With Accela filter on, no movement at all. Only with no filter it is moving. When I have found software to record both in-game and desktop I will record some video. I previously downloaded a record tool which was full of spyware, almost killing my Windows installation.

sthalik, 2015-07-03:
Is the octopus also stuttery? Also what's the deal with Accela?

sthalik, 2015-07-03:
Try this one: https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll

It uses a different approach.

MathijsG, 2015-07-03:
No the octopus is totally fluid, just like the game when the game isn't on focus. The accela problem is new to me, even then the octopus is moving. With EWMA i have the previous problem of not seeing an octopus at all (refering to: https://github.com/opentrack/opentrack/issues/169)

sthalik, 2015-07-03:
The same with dropbox binary?

MathijsG, 2015-07-03:
I will try that one again, after the problem with the empty profiles I deleted everything of the previous builds and started fresh with this build, so I can exclude everything while testing.

sthalik, 2015-07-03:
On another note, does newest Accela perform significantly worse?

MathijsG, 2015-07-03:
With the dropbox version I can't even get any movement at all anymore, weird. I will try the new Accela, the use of EWMA was just a testing purpose. My previous 'problems' with Accela were that the movement felt a bit like a stair when you draw a diagonal line with your head, instead of fluid natural motion like FaceTrackNOIR had with these filters. But I will test the new one out :-).

sthalik, 2015-07-04:
Accela tries to perform fast a movement that's over 2.5x the threshold. Maybe we're overdoing it, and they should be dampened somewhat.

MathijsG, 2015-07-05:
With the new version I get this error again:

[image]

Previously I only got that when I clicked the old Flight Simulator profile. It's automatically crashing after that message.

sthalik, 2015-07-05:
It doesn't crash for me after the message in development build. There was an error when it was assumed it failed even if it worked. Also now it works even if started before FSX, or if FSX is quit and start again.

Confirm this build works, and I'll release p2. https://www.dropbox.com/s/5gyh8q02nm0rcby/opentrack-2.3-rc13p1-10-gf8370ea.zip

MathijsG, 2015-07-05:
I finally made a video :-). With the newest version of your Opentrack the tracking itself is working again without crashing, but the stuttering still is alive. I hope the video clarifies my story: https://www.youtube.com/watch?v=P8mxX1AIMkk

sthalik, 2015-07-06:
I'm unable to reproduce it on a clean install of FSX + Acceleration + SDK. FSX is a hog in general but I don't get a dramatic slowdown like you do. Can you try to force-disable vsync and/or do a clean install without mods?

On another note, is Accela in that build worse than p1? I understand FSX doesn't work for you but maybe you have other feedback.

MathijsG, 2015-07-06:
I doubt it has anything to do with graphical settings. The game reaches
100+ FPS and as you can see further in the video facetrackNoIR works
perfectly. Before I had regular FSX but this is the steam version called
Steam edition, I will try some further testing. It's so weird that if the
game doesn't have the focus it's working perfectly with both tracking
programs.

sthalik, 2015-07-06:
Replace this: https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll

Facetracknoir working gave me an idea.

MathijsG, 2015-07-06:
Still stutters while having focus :(. Although it *seems* a little less stuttery, but I should look more into it later.

sthalik, 2015-07-06:
Facetracknoir doesn't give itself high priority. How many cores you got? With four it should be fine. PT uses its own fair share of CPU usage. If you got two it may show why I can't reproduce.

We can use SetThreadPriority on either SimConnect thread or PT but which one? That needs some round-trips between you as tester and the binaries.

MathijsG, 2015-07-06:
I have 4 cores.

i5 2500
16GB Ram
7950 3GB

Must be enough :). I'm glad to help out.

sthalik, 2015-07-06:
OK try https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll

It sets thread priority ABOVE_NORMAL. There's also HIGHEST. We can also decrease the timeout if it works.

sthalik, 2015-07-06:
DLL ninja updated. See 3b70e63

MathijsG, 2015-07-06:
Still stuttering while having the focus, and perfectly smooth when out of focus, with this protocol: https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll

sthalik, 2015-07-07:
Link updated, https://www.dropbox.com/s/r6yvelvp517t580/libopentrack-proto-simconnect.dll

It's using high priority (maybe requires admin) in addition to event handle as previously.

MathijsG, 2015-07-07:
With the new DLL it doesn't show up on the plugin list here.

sthalik, 2015-07-12:
@MathijsG confirm in rc13p2. High priority added.

MathijsG, 2015-07-15:
Hi, tried rc13P2 today. Unfortunately still stuttery until the game doesn't have the focus. Weird problem :-S.

sthalik, 2015-07-16:
Can you check if it stutters too with joystick tracker? We need to figure out which thread needs its priority bumped.

MathijsG, 2015-07-16:
I will check today.

Two questions:

- Do you have any idea why it does work already when the game hasn't got the focus?
- What is the difference in protocol between the version in FaceTrackNoIR which works flawlessly?

sthalik, 2015-07-16:
- game detects focus and does something differently.
- it calls the game only 66 times a second. other than that, no idea.

MathijsG, 2015-07-16:
Although i'm instead of viewing around, controlling the plane with my head with vJoy it is fluid at the moment.

sthalik, 2015-07-16:
That's not what I meant. Use the joystick tracker with simconnect protocol.

MathijsG, 2015-07-16:
This seems to work smoothly.

sthalik, 2015-07-16:
ftnoir calls 66 times a second, we call 250 times a second.

If tracker joystick works fine, then it's PT thread that needs adjusting priorities. I'll have a test build soon.

sthalik, 2015-07-17:
Open PT dialog window, see FPS. Do you get lowered FPS when FSX has focus?

sthalik, 2015-07-17:
Open Task Manager. Go to CPU, switch view to per-core. Compare the first core's CPU usage when FSX is in focus and not in focus.

MathijsG, 2015-07-18:
Checking it now. Results so far:

Significant FPS drop while focus or reverse? No
Significant processor usage difference while focus or reverse? No
PT tracker FPS drops significantly though. While not focussing FSX it's the regular 75 FPS of my PS3 Eye Cam. When focussed it's dropping to 23-30

No focus:
[image]

Focus:
[image]

sthalik, 2015-07-18:
Try this build https://www.dropbox.com/s/m1jmgnw9h7lhqdz/opentrack-test-no-affinity-20150719.7z need to get it out of the way. FPS should stay at 75.

MathijsG, 2015-07-18:
Nice work man! Working perfectly again!

What was the fix in the end?

By the way is it a good approach to replace everything except the "settings" folder when updating? Because otherwise I would overwrite my default profile.

sthalik, 2015-07-18:
> What was the fix in the end?

We used to set thread affinity to cores 0 and 1 for PT and pipeline, respectively. The build you tested had no thread pinning.

> By the way is it a good approach to replace everything except the "settings" folder when updating? 

Store profiles elsewhere.

commit 64a6892e (2015-07-18): don't set thread affinity for anything

Fixes #174
closed 2015-07-18 by sthalik