opentrack/opentrack #176: Dynamic resolution bug or improvement

rizer1980, 2015-06-21:
Hello.

When one of the points lost, the red dot moves to the wrong place.
Coordinates are no longer properly transmitted.
It does not help either reset or switching on / off via hotkey.
Just stop and start through the application.

Video:
http://youtu.be/YEWlZ01GALw

I have two suggestions:
Or do so would by pressing zero, dynamic resolution also stop start.
Or do as freetrackNoIR, with the setting the "reset time Dynamic resolution" when I put there instead of 1000ms, 100ms, it works well.

Thx.

sthalik, 2015-06-21:
Right, dynamic pose resolution timer needs to return badly.

sthalik, 2015-07-03:
Assumed fixed in 95218d013693e3af637fe4ea3333090b8eaa0500. If still broken I'll reopen.

closed 2015-07-03 by sthalik