opentrack/opentrack #177: NPClient.dll Gets blocked by BattlEye ArmA 2; No TrackIR scheme in settings ArmA 3;

Suppresed, 2015-06-25:
Hello

I just dusted off my ps3 eye camera and my 3 led headphone piece to procrastinate a little, so I went ahead and looked for the newest OpenTrack (2.3rc12p1), as I deleted the old one (2.2stable), I opened ArmA 2 and OH DAMN! There was this little CMD telling me it have had blocked NPClient.dll, With a little bit of thinking I downloaded the old one (2.2stable) and tried one more time... IT WORKED!
The thing worked properly, but then I thought it will be a shame all the new features (better design) of the most recent one (2.3rc12p1) being thrown away just because the thing got blocked.

As I supposed replacing the new dll's with the old ones as the old one weigh 16KB and the new one 19KB... The point is that I tried and it worked, so now I know the goddamn dll is the issue, is it ok to use the old dll's with the newest version? Cause my mouse kinda lagged a little when using this mixture.

Any help will be very much appreciated!

Thanks!

PS ArmA 3 does not show the TrackIR controller scheme in settings when using either of the dll's. Maybe a game error... Downloading again. Thanks again!

Suppresed, 2015-06-25:
Happy to see nobody cares

MathijsG, 2015-06-26:
What settings do you use in Freetrack profile? I just emulate the TrackIR, never had to do anything with DLL's.

sthalik, 2015-06-26:
Their anti-cheat system indeed blocks our npclient.dll. There's nothing wrong on our end, but I'll try reaching out to Bohemia Interactive if they can whitelist our DLLs. I need a breka however and it's gonna take some time. Take care.

sthalik, 2015-07-01:
"Fixed" in a030d0281589fc1e3490981a0ed4207336c62b32

closed 2015-07-01 by sthalik

MathijsG, 2015-07-01:
Can you technically explain what was "wrong" and why they blocked it and not block it anymore after your fix? Just interested :).

sthalik, 2015-07-01:
I reverted the .dll to version from 2.2. Locking changes make it read and write data at the same time with no mutual exclusion. It should work fine, floats are 32-bit so reads should be atomic.

sthalik, 2015-07-02:
I wonder if DCS has problems anymore with the changes. It's possible that lack of mutex usage fixes the "flash to center" single-frame issue.

sthalik, 2015-07-03:
@Suppresed @MathijsG believed fixed in RC13.

sthalik, 2015-08-21:
For what it's worth Arma 1.50rc works with both freetrack and npclient now.