opentrack/opentrack #180: [bug] Default dimensions for clips and caps are all zero'd in Pointtracker plugin

MathijsG, 2015-07-03:
Normally this plugin would have the default dimensions for clip and cap in the software. I think with this all zero'd it would make the software harder for users.

[image]

[image]

sthalik, 2015-07-03:
It can't be any better. If dimensions are half-right by default, users will leave them there and complain that tracking works erratically.

MathijsG, 2015-07-03:
But before they were based on the defacto default which anyone uses. Why differ from that? The few people with different dimensions will change it, but the majority with a default clip now find their program not working unless they fill in these dimensions somewhere hidden in the program.

sthalik, 2015-07-03:
What default? Please provide a reference and back up how "everyone uses it".

MathijsG, 2015-07-03:
In the Freetrack Manual: http://forum.free-track.net/index.php?showtopic=413

Every program or software I saw which involved home made clip making defaulted to:

[image]

I thought FaceTrackNoIR and even the previous versions of OpenTrack also had these dimensions pre-filled in. Or am I mistaken?

sthalik, 2015-07-03:
Fixed in 7a98f8831efb6692241106e53af09e2385f02668, thanks!

closed 2015-07-03 by sthalik

MathijsG, 2015-07-03:
Maybe also insert these 'default' hat dimensions?

[image]