opentrack/opentrack #184: [bug] Accela dead slow even with all sliders to lowest point

MathijsG, 2015-07-03:
Somehow the Accela filter has become dead slow and has a big movement delay. Before it was almost as responsive as no filter with the sliders to the left. I definitely need to find some decent recording software to show you what I encounter.

sthalik, 2015-07-03:
Do you have deadzone or smoothing enabled? Does it happen if you only play with "rotation sensitivity"? Accela changes aren't final so it probably can be fixed.

Len62, 2015-07-03:
I noticed this also, all accela settings at zero, lots of lag time when moving. Dead zones in curves practically at zero. I'll do some more checking.

MathijsG, 2015-07-04:
I have them like this:
[image]

And even then it has a huge delay. You don't encounter the same Stanislaw?

sthalik, 2015-07-04:
I believe it's fixed in 2e418fc05c87a1e0754c2a72556f5416ccaf6116. This warrants rc13p1.

It should work like Accela mk2, except the user controls a slider and not curves. Works better for me than rc12's "staircase" version.

MathijsG, 2015-07-04:
I will try it today. Is there already a difference between when you select accela without enabling any of the sliders to more than zero, and no filter at all?

Len62, 2015-07-04:
That fixed the issue for me. Thank you!

closed 2015-07-05 by sthalik

severgun, 2015-11-26:
Actually it is still slow.
http://www.youtube.com/watch?v=j7WSBAb9RrU

sthalik, 2015-11-29:
@severgun is the maximum velocity an issue in practice? It was balanced to keep up with realistic head movement speeds.

severgun, 2015-11-29:
@sthalik you can see head movement speed as it is at edtracker window. Yes, it is quite fast motion. But such move is often used in flight simulators in dynamic battles. For example, fast roll and you need to move head side to side. If sensitivity is 1.0 it is OK. But if you set sens to 3.0+ you will get slow reaction.
Actually I'm using such high sensitivity because there is no zoom on curves windows I can't set curves accurate on first 20-30 deg.

sthalik, 2015-11-29:
You're not meant to use sensitivity way higher than the precision of the tracker. You need 3 times as much change to achieve the same velocity in that setup.

You should be able to get along with logarithmic mapping for the first 30-50 deg by merely setting a few points, and the graph will change the shape. You needn't dictate every few degrees with control points. It's enough to use four total for the whole mapping. With deadband my yaw mapping was fully linear BTW.

I don't understand the issue by now. Is it low maximum velocity or sensitivity response by now?

severgun, 2015-11-29:
> I don't understand the issue by now. Is it low maximum velocity or sensitivity response by now?
Yeah my english not so good :)

If I set high sensitivity I can use all width of curves window and make something like on top part of image, but then I get slow not usable accela.
If I set sensitivity to 1.0. Accela working fine, but I have to setup curves only on 20-30 deg. like on bottom part of image. Without zoom it is quite tricky. 

http://i.imgur.com/OV8YumR.png

What I want? Fast accela at high sens. Or maybe zoom at curves setup window.

PS Just find out that accela causing flickering at low smoothing. Should I create ticket or this is normal?

sthalik, 2015-11-30:
If you use less of the mapping window then effectively you increase Accela's top velocity.

I'll consider adding another velocity step after the existing ones. The existing ones were tuned for noisy webcam trackers and looking through a scope in Arma 3. The new step will be added somewhere after the current limit - 2.6 times the sensitivity in degrees. I won't touch the existing steps.

sthalik, 2015-11-30:
Do you need any filtering at all though? Does it work with no filter?

severgun, 2015-11-30:
Yes it work without filter, but of course my head shaking little bit.

sthalik, 2015-12-02:
@severgun compare https://db.tt/jia1zt3R

Velocity gets higher for even higher movement deltas.

severgun, 2015-12-02:
I did not test well but...
Flickering Game Data on low smoothing fixed. Accela response fast.

3 times it stoped tracking after ~2-5 min and If I set Yaw axis to make fake X axis I get TZ game data recieved.

sthalik, 2015-12-02:
What do you mean stopped tracking?

severgun, 2015-12-02:
Button Start pressed and instead of octopus there was pink pixel noise. Will test it more.

About Z axis that was my fault. I forgot to mark checkbox to disable Z compensation.