opentrack/opentrack #185: •PT point extraction method was changed.

Len62, 2015-07-04:
In previous versions I had to keep the threshold almost completely to the left side. With the new version I can set it just below halfway. Overall, using 3 pt clip, PT, and accela filter RC13p1 is working great!

sthalik, 2015-07-05:
I'm happy to hear that.

There's one last Accela change not in p1 - moving fast left/right makes the view jerk too fast. I've set it to a minimal delay to avoid that unnatural feeling.

How is Accela in rc13p1 working for you guys? Name-dropping @MathijsG @asarnow @FlyingCircus- @Trooph @forteh :-)

rizer1980, 2015-07-05:
I seems better than all versions before!
Thanks.

But in 2.3.12 p1 you remove Dynamic pose timeout, again.
It was in 2.3.12, but i have problem with delay with accela filter like here:
https://github.com/opentrack/opentrack/issues/184

If it is any problem with this parameter, can you add hot key to reset dynamic pose?
It is only one thing from ideal tracker for me)
When the battle is lost in the red dot, and I have in the midst of battle to switch to windose that would push stop-start.

sthalik, 2015-07-05:
@rizer1980 there's dynamic pose timeout in the published build. You might have replaced the PT .dll with something else. Try redownloading the zip.

You can set a timer up to ten seconds. Trackers shouldn't have their own hotkeys.

MathijsG, 2015-07-05:
It's working very nice :). Although I do encounter a little less responsiveness on accela versus no filter, but I think that's by design. Will test further.

FlyingCircus-, 2015-07-06:
@sthalik I recently moved to a different city and I didn't bring my IR clip, therefore my lack of feedback lately. I'm in the process of building another one, I'll share my impressions soon

sthalik, 2015-07-12:
Can you use Accela now without smoothing and deadzone? What's your config?

MathijsG, 2015-07-16:
It feels very fluid with WarThunder for me. Just need a little deadzone because my clip has some after movements after moving. This is my config:

[image]

Still playing a little with the settings, for me these values were quite arbitrary.

sthalik, 2015-07-16:
We should probably reduce slider max values since no one will use that much.

Len62, 2015-07-16:
RC14 performing great! These are my accela settings:
[image]

FlyingCircus-, 2015-07-19:
@sthalik I finished my new IR clip. New accela works perfectly after tweaking the settings again. I'm surprised I could use very small deadzones in comparison with older builds. I also noticed I could lower smoothing to barely nothing and gain more responsiveness than I had before. I miss a joystick now :(

FlyingCircus-, 2015-07-19:
@sthalik There's something odd happening: I get TZ and TY variations in the game data even if I see those coordinates are in the flat part of the curves:

[image]

In the example you may notice TZ = 17 while it should be 0. This is causing some cross axis behaviour.

sthalik, 2015-07-19:
You have translation compensation on.

FlyingCircus-, 2015-07-19:
Oh, yes, I forgot about that one. Everything is fine then

sthalik, 2015-07-23:
Did rc15 internal change break tracking for you guys? It simplifies the code, but can break pitch output and model pitch value significance.

Len62, 2015-07-23:
I'm out of the loop until around the 31st. Miss not being able to test this for you.

FlyingCircus-, 2015-07-23:
@sthalik I could barely test it today but, as you said, model pitch was completely off (red cross was way higher than it should be) and rotations were also uncontrollable. I couldn't get to the bottom of the issue, it wasn't only that the sign was inverted, it was something else

sthalik, 2015-07-26:
For the record rc15p1 reverts the breaking change.

sthalik, 2015-07-28:
Believed to work.

closed 2015-07-28 by sthalik