opentrack/opentrack #186: [feature] Make tracking start when it isn't while clicking model position calibration

MathijsG, 2015-07-05:
Currently you can't calibrate while the tracking isn't starting, you need to close the window again, press start and open the window again.

Maybe starting the tracking while clicking on the 'Toggle calibration' button would be a more noob friendly solution?

[image]

sthalik, 2015-07-05:
Works for me, check rc13p1.

sthalik, 2015-07-05:
That is, calibration button becomes active when tracking starts.

MathijsG, 2015-07-05:
My point is, make this button clickable and when tracking isn't started yet, start it when clicking this button. Instead of first closing this window, going back to the main program, click start and then opening this window again :).

sthalik, 2015-07-05:
You don't need to close the window, button state updates when tracking starts.

Trackers don't have access to start/stop-related functionality. It can't be done from outside the executable module. The responsibilities are decoupled on purpose, so the code is less complex.

closed 2015-07-05 by sthalik