opentrack/opentrack #187: [Feature request] Make a button or a window for specific webcam settings

MathijsG, 2015-07-05:
With my PS3 camera I use a special software tool called CL-Eye Tool, but for many webcams you can't easily alter settings of the webcam without some program like skype or something which can start the webcam settings, which can trigger this window:
[image]

Maybe OpenTrack should also have some settings button, so you don't need external programs to tweak the webcam view. FreeTrack also had this in their program.

sthalik, 2015-07-05:
I agree. OpenCV has an option to pop up the dialog, but in case of PS3 Eye it doesn't work, even though it works in Skype.

Working with DirectShow is overkill for me. Someone else has to write the code.

sthalik, 2015-07-06:
Implemented. Won't work for PS3 Eye.

Doesn't work while tracking for HT at all. Either HT stops being a subprocess or it won't work.

sthalik, 2015-07-06:
We could temporarily stop tracking for HT though.

closed 2015-07-06 by sthalik