opentrack/opentrack #195: [Feature] See current pixel size

MathijsG, 2015-07-19:
I sometimes get annoyed by random pieces of sunlight reflecting in the webcam. Pttracker picks this up as another reference point and the tracking stops because it sees 4 dots. Very often I can block this sunlight with the treshold slider and with the correct sizing of the points, however it is a little hard to see the actual pixel size of your points at the moment:

[image]

To me it seems handy to have the current minimum size and maybe maximum size of the 3 points somewhere in this above window, so you can easily limit the size. Now it's just a matter of trial and error to get the correct range.

sthalik, 2015-07-20:
It might be an annoyance but addressing every such annoyance makes it worse. The UI becomes cluttered with extra options and labels for very rare purposes.

Point extraction is weird now, it samples at several similar thresholds and only outputs the point if all samples succeed. It's also closer to the center the more circular it is in given sample. It might output "blue" for thresholded but won't show as a point since higher sample(s) failed.

sthalik, 2015-07-20:
Maybe display it on the image itself?

MathijsG, 2015-07-20:
That can be a nice solution. It's not high prio for me, it's just making the program a little more userfriendly to me, because you only see the big crosses and no dots itself I just have to guess the pixel size.

sthalik, 2015-07-21:
You can always move the threshold slider to max value and then the points will show.

Crosses can be changed to dots surrounding the point with some margin.

Issue with putting text on the video feed is that it's downscaled a lot, but not divided by two. Text always looks blurry and aliased like in Aruco tracker display.

MathijsG, 2015-07-21:
The text in Aruco webcam window seems good enough for me :).

sthalik, 2015-07-21:
Further, you can adjust min/max size until it no longer detects the point, with no need to apply settings, as they take effect immediately. Is that not good enough as well?

BTW I have an idea how to sort points better by angle relationships. Maybe it can reduce the need for dynamic pose resolution.

MathijsG, 2015-07-21:
That's how I do it now, it's not really a problem now, but I thought it would be more userfriendly with actually displaying pixelsize somewhere. But maybe we shouldn't clutter the program too much either.

sthalik, 2015-07-22:
Dynamic pose helps the tracked see which point is which on extreme yaw and similar. It can guess when reasonably centered, only.

People still complain about opentrack being complex to configure and get around. Actually I'm doing a spinoff for one particular clip/cap manufacturer with most of the stuff simplified. All that doesn't pertain to their devices was changed AND the software is still a bit complex when first started.

If we had reasonably necessary information about the points, text can be introduced. There's some valid data to be put there, like reprojection error. But I can't think of anything that even 20% users want in this form.

sthalik, 2015-07-22:
Can we start a brainstorm for a wiki page with easy-to-ingest first-time-user information? Maybe the software should FINALLY get a configuration wizard, covering common use-cases like PT and Oculus Rift. If we point users in the right direction by small bits, it's gonna work. See my failed attempt at https://github.com/opentrack/opentrack/wiki/concepts-used -- no one's gonna read that kind of thing.

closed 2015-07-22 by sthalik

MathijsG, 2015-07-22:
I would gladly be of help, and maybe even offer some Skype support with helping people setting it up. I did this before on a computer forum. A configuration wizard isn't a bad idea at all. Or maybe some explain buttons or tooltips to have the more abstract things like 'dynamic pose resolution' explained. I sometimes miss that and need to google things manually, which is okay but could be better.

sthalik, 2015-08-24:
@MathijsG implemented for rc17.