opentrack/opentrack #196: Unclear what is stored in profile and what globally

MathijsG, 2015-07-21:
With setting and tweaking up my profiles I encounter that things like key shortcuts and which webcam to use is gone when I pick another profile. Can you clarify which settings are now saved in profiles and which are globally saved?

Maybe in future something like "Apply to all profiles" seems handy to me?

sthalik, 2015-07-22:
Everything except game detector data and the current profile is stored in .inis

Apply what to profiles? Given module's data like selected filter's?

It depends on some things:
- reasonable profile management, like storing them in a predefined place. it's been far too long in its present shape.
- removing checkboxes from modules' dialogs AND exposing their settings through an interface, so that the main executable can understand what to apply in the first place
- hatire needs to be converted into the new profile framework
- profile framework needs to be extended for being used by the mapping editor, otherwise it can't be applied the same way.

MathijsG, 2015-07-22:
For example, camera settings on the pointtracker is a global setting to me, meaning that it wouldn't differ per profile in my opinion. Same for the camera offset settings at the settings window.

sthalik, 2015-07-23:
That's more complex. PT treats its settings as one bundle, over all the tabs.

sthalik, 2015-07-26:
We'll definitely want to embed protocol/tracker/filter settings in tabs of the options window, but not separate them into independent bundles for .ini purposes.

sthalik, 2015-07-28:
@MathijsG what can we do with it? Going after very single module (tracker, filter, protocol) and splitting its bundles into multiple ones seem like the only solution. But then these bundles need to be exposed into UI with enough information for the user to choose which bundle to apply.

sthalik, 2015-07-28:
We need though to move tracker/filter/module settings into a unified "options" window, as tabs. This should help ergonomics.

sthalik, 2015-07-28:
We should default to a face tracker, but only after improving it beyond its present pitiful state.