opentrack/opentrack #199: Reference point is on a totally different spot than same profile on older version

MathijsG, 2015-07-23:
I'm trying out the new RC.

Old program:

[image]

[image]

New program:

[image]

[image]

With the reference position of my head totally wrong in the new build, based on values which worked in the previous version, I'm unable to test headtracking because it's way off now and the rotation is all wrong. Calibrating it makes it only worse. Did something change with this?

sthalik, 2015-07-23:
Please test it like so:

- put your model upside down
- wiggle it

Does it match what happened with old rc in the normal non-upside-down position?

sthalik, 2015-07-23:
The issue is only for pitch. That's the part that changed in internal representation.

sthalik, 2015-07-23:
Alternative you can switch 40 to -40 and 70 to -70 and it could help.

MathijsG, 2015-07-23:
But the visual location of me (I think that's the purpose of that red cross?) is also totally different, and with rotation and translations it picks a wrong reference then.

I will try the above things asap.

HMRMike, 2015-07-23:
Same here. I set up everything with the same values and curves as I used in RC14.
The new coordinates for calibrated model position seem very wrong, placing the point of origin somewhere on my throat. Large +TZ and +TY output when I pitch up. But barely any reaction (in any axis) when I pitch down.

Flipping the marker upside down and calibrating seemed to produce a more correct output, but I can't test the accuracy yet.

MathijsG, 2015-07-23:
I can't test this properly upside down unfortunetaly, although it seems more correct upside down. The calibrating is just way off.

MathijsG, 2015-07-23:
With the -70 and -40 trick it's working more okay. Will test further in-game.

Daggett57, 2015-07-23:
This is happening to me too. I get yaw and roll interconnect and the auto calibrations gives wonky number like 820 mm for the y model offset. Inverting those two numbers seem to help a lot, if not fix it.

sthalik, 2015-07-24:
I've reverted the change. In fact it was only needed for calibration, which was also removed, so it was reverted without a hassle.

Please confirm if rc15p1 works. Thanks!

MathijsG, 2015-07-24:
Why did you kill calibrating by the way? How are people supposed to use Pointtracker without a form of calibration? It was working for me in previous builds?

sthalik, 2015-07-24:
I didn't kill it. in rc15p1 it should work fine.

sthalik, 2015-07-24:
Are you talking about camera calibration? The values in data files were all wrong. For instance focal length was too similar with 56/75. To do camera calibration I need more research and confirmation, rather than pushing the change on unsuspecting users.

MathijsG, 2015-07-24:
Looks like it is working again. Calibration still puts the red crosshair on a different spot, but looks like it is working. Only problem with some yaw when I only move up and down. Will dive further into it. Overall very smooth.

sthalik, 2015-07-24:
Calibration isn't absolute, i.e. it differs basing on exact sequence of data it receives.

Please report further issues if they arise.

closed 2015-07-24 by sthalik