opentrack/opentrack #200: [UX] Axis inverting settings too hidden away.

MathijsG, 2015-07-23:
Many games interpret the axis differently. When you play many games you often need to invert some axes for certain games. In my opinion this important functionality is too hidden away somewhere in a tab  titled output at settings. Maybe this important feature should receive a more prominent location? Perhaps even on the main window like FaceTrackNoIR had?

What's people's opinion about this?

sthalik, 2015-07-23:
I've changed all the external modules' interpretation of +- to be the same. That involves trackers and protocols. If a game-specific protocol misinterprets the sign, we need to fix that game-specific protocol.

sthalik, 2015-07-24:
If you know of games that misinterpret the signs, please share.

MathijsG, 2015-07-24:
I will. I had to reverse my pitch in FSX steam. But how can you make this universally correct? Some people have maybe webcams that automatically mirror the webcam image and stuff?

sthalik, 2015-07-24:
Can you confirm this:

- freetrack protocol with same tracker, no need for reversing
- simconnect with same tracker, reversing needed for pitch

If so, I'll fix it.

Webcams that mirror have a mode that's for picking one's nose and one "real" mode. Worst case, they indeed need reversing. But it shouldn't be necessary or visible for most users.

MathijsG, 2015-07-24:
Hmm, tried FSX and Warthunder with default profile with no adjustments to the axes and they are both working perfectly. Now I try my own Flight Simulator profile with inverted pitch again, and now I need to re-invert in (disable the check button), Weird, but it's working fine :). Or did you just recently change the behaviour?

Anyways my games are working perfectly in order of axes. When I encounter a game where I need to invert the image I will kick this issue up again.

Thanks!

sthalik, 2015-07-26:
I changed it while we were debugging the stutter with FSX in focus.

I'll reopen when we have another problem.

closed 2015-07-26 by sthalik