opentrack/opentrack #202: No OSX Plugin

bodylinksystems, 2015-07-24:
Default protocol is FlightGear on OSX. Currently no way to get data into XPlane for head tracking but Android app/OSX build combo is working fine outside of the game.

sthalik, 2015-07-24:
Presently the X-Plane plugin is built only for Linux. I'll try to get it inside OSX.

bodylinksystems, 2015-07-24:
I'm happy to test if you need help.

sthalik, 2015-07-24:
This is the X-Plane plugin untested version: https://www.dropbox.com/s/7cntkelhg018e1a/opentrack.xpl

The binary only works with 64-bit X-Plane. If there's only 32-bit X-Plane then we have to make changes.

I can't test it directly due to lack of OSX with connected real GPU.

sthalik, 2015-07-24:
Oh, and use Wine protocol with X-Plane even thought it might not make sense.

bodylinksystems, 2015-07-24:
Will test tomorrow on 64-bit x-plane, choosing wine protocol.

bodylinksystems, 2015-07-24:
Did a quick test and the plugin works like magic, however I get alot of screen tearing on all current settings of acela. This happens moving along both the x and y axes. Roll does not seem to work at all.

Can map toggle on/off to joystick button with success, however re-centering and zeroing behavior is not intuitive or not working. Need more testing.

Tested as such:  Created folders: plugins>opentrack>64 and renamed plugin 'mac.xpl' to match other plugins

Overall, this works quite well but the screen "tearing" during movement is a real killer. Other headtrckers I have tested (linuxtrack) did not have this behavior.

Thank you, and looking forward to continued testing. Will upload videos if that will help development.

sthalik, 2015-07-24:
Roll isn't supported by X-Plane with the API we're using. We could use the same method linux-track uses but it's very low-level. I don't think it's appropriate to need to do what linux-track does so that method was ignored so far.

Some keys on the recenter/toggle/zero list may not work on OSX. You should try until one of them works. This will change before release.

I'm putting up an experimental build to see if it has anything to do with tearing. https://www.dropbox.com/s/7cntkelhg018e1a/opentrack.xpl

sthalik, 2015-07-24:
X-Plane is using now pilotview plugin. Should we do the same? This is only relevant for roll input.

bodylinksystems, 2015-07-24:
dziękuję! This is so fantastic!

Affirmative. I don't miss the roll aspect in xplane, just thought I would mention it for testing.  I might also turn off the headshake plugin for better testing?  I should also note that there were only 2 options for opentrack key bindings in x-plane : Toggle and Toggle Translation. Not sure what toggle translation is?

Will try v2 of the plugin and report results this evening. Focusing on finding center/toggle/zero keys and tearing.  IMO, I would not like keys for these things, I would rather bind them to a spare button on my stick.

sthalik, 2015-07-24:
These keybindings function inside X-Plane and control the plugin. Centering and the like are performed inside opentrack, which hijacks the key and makes it unusable until the key's changed or tracking's stopped.

I've started working on binding arbitrary keys.

But don't count on joystick support for Mac, anything for Mac requires Objective C, so far I've avoided writing any.

bodylinksystems, 2015-07-25:
No problem. I can bind the keys in x-plane and toggle works pretty well, with the added bonus of a re-center when re-starting -- so the way I am using it is for the toggle and for centering and zeroing. It works for me.

When I stream, I show the outside views quite a bit because I make custom liveries for my JS32

bodylinksystems, 2015-07-25:
Did some decent testing tonight. Here is what I know so far:
the second plugin you provided did not work at all, so I tested the first a little deeper.

Couldn't find consistent behavior with the toggle /center/ zero buttons assigned in opentrack. Just didn't seem to do anything, then at times it seemed to do something but I never know what "mode" I was in -- if that makes any sense.

Binding a joystick button to opentrack/toggle was *very* nice, but it behaved sporadically. However when it did work as intended, it was perfect. toggle it off, do some things, kick it back on, tack the head. It was so nice.

the actual tracking was still fairly jittery and there was some tearing. However i get that with quick movements of my mouse as well. I just feel that opentrack needs more smoothing. It feels like a sawtooth curve instead of a smooth curve. I tried to tweak the settings, but jittery behavior was persistent.

I mounted the phone on my head very securely via a custom mount I made and think using some sort of 1/4-turn mount like a Garmin Edge 500 would be best.

sthalik, 2015-07-25:
Hey,

Is it screen tearing or maybe you get noise that doesn't get filtered? The view should stay still when you don't do anything.

OSX support for shortcuts should now allow for binding arbitrary keys. It's going to be in the next release.

sthalik, 2015-08-21:
Closing abandoned, sadly. Reopen if you get back.

bodylinksystems, 2015-08-22:
I'm back and able to do a small bit of testing with the latest build (2.3rc16) and x-plane on 64-bit. Let me know where to focus my testing to best help the project.

I should note that my joystick assignments are now completely messed up and it seems they are playing "whack-a-mole" with me. I set one up to be "button x" and it goes to "button z". I have some funky things going on with my joystick buttons now.

sthalik, 2015-08-25:
Head tracking is meant to be used from cockpit view. Does it work then?

Do keyboard shortcuts inside opentrack work now? It should intercept all possible keys.

You can use in ETS launched from Wine, using Wine protocol. Just run opentrack before just once with same WINEPREFIX. Otherwise, there's no virtual joystick support for OSX.

bodylinksystems, 2015-08-25:
Will test key bindings, but I have so many key bindings in x-plane that I didn't know which one to choose!  I will also try with ETS2 and report back.  Thanks again for the awesome work.

sthalik, 2015-08-26:
@bodylinksystems to be clear, report whether opentrack keybindings from opentrack settings window work, first. I'll look into X-Plane keybindings later.

bodylinksystems, 2015-08-27:
I can report that keybindings do work from within opentrack. In fact, the opentrack window does not even need active status for the key bindings to work. The only issue is the joystick binding of opentrack needs to be pressed about 3+ time to disengage opentrack.  Other than that. I can report that it works in x-plane 100%. Will test in ETS2 as well.

sthalik, 2015-08-28:
Does it still need to be pressed multiple times if selected from menu? Also, are you able to run X-Plane from terminal to gather debug info?

bodylinksystems, 2015-08-28:
I understand what you are asking and it's hard to untangle what is a joystick button issue versus the opentrack key bindings. I will have to unbind the joystick bindings in x-plane and then try it entirely from opentrack.

I have never executed x-plane from command line, but am happy to do so. I would just need to make sure that I am calling the 64-bit version from the command.

sthalik, 2015-08-28:
I'm confused now which keybinding it is. Are you using keys that are bound in X-Plane or these in opentrack?

sthalik, 2015-10-11:
It's been a while without @bodylinksystems response. Closing.

closed 2015-10-11 by sthalik

bodylinksystems, 2015-11-10:
sorry, I tested this and it works perfectly. I have been playing alot more DCS on PC. This solution works perfectly on x-plane 10 for osx